foxglove/studio #3119: Visiting a url with a layoutId that is not yet loaded (from remote sync) doesn't work.

defunctzombie, 2022-03-31:
If I make a layout and send a link to the layoutId to a team member, that layout does not load for them if they haven't sync'd their layouts.

Layout loading should work even if there hasn't been a sync.

foxymiles, 2022-04-18:
What approach should we take here? Force a sync and then re-try to load the layout? Should we have some kind of modal state for this?

defunctzombie, 2022-04-18:
My initial thought is that the CurrentLayoutContext/Provider needs to be enhanced to support a "loading" state. I believe right now we have a binary state: no layout and layout. With no layout we show the "select a layout from the sidebar" fallback, and with a layout we show the layout.

We need to update the state to a type that can represent four states: loading, no layout, layout, information. Now when we have the layout manager load a layout by id, we can update the layout context to indicate loading while the manager loads the layout. If there is no layout to load, then we use the no-layout state. If we have a layout we can show that, and finally if there is some permission error or load failure we can show that too (and maybe offer retry).

The **tl;dr;** is that we need to enhance the notion of the "current layout state" beyond binary present or not preset.

PR #3329 (2022-05-06): Add remote fallback to LayoutManager#getLayout
**User-Facing Changes**
When a user loads a url with a layoutId, Studio attempts to load the layout id from their account (if signed in). This creates a better link-sharing experience when working with team shared layouts.

**Description**
When a user visits a studio url (or deep link) with a specific layoutId, the expectation is that the app will display this layout to the user. The app attempts to load the layout by asking the LayoutManager to getLayout with the layout id. The layout manager consults its local cache and returns the layout with t [... cut]
commit 77aa83a7 (2022-05-06): Add remote fallback to LayoutManager#getLayout (#3329)

When a user visits a studio url (or deep link) with a specific layoutId, the expectation is that the app will display this layout to the user. The app attempts to load the layout by asking the LayoutManager to getLayout with the layout id. The layout manager consults its local cache and returns the layout with that id.

When working in a team setting, a user may receive a link from a co-worker with a layout id. When the user clicks this link, studio attempts to open the layout id but is unable to because the app has not yet sync'd the use [... cut]
closed 2022-05-06 by defunctzombie