foxglove/studio #3122: Save URDF as part of layout

drldcsta, 2022-03-31:
**Current Options**
Currently, urdf models are only able to be displayed by advertising a native ros topic or having a local `package://` declaration. Both of these don’t work with out setup, as we aren’t using a native ros connection, and the `package://` requires us to use the desktop app. 

**Proposal**
We’re proposing that the URDF file be saved as part of the layout in foxglove. This would look like a file upload option on the front end.

jtbandes, 2022-03-31:
This wouldn't be as simple as saving a _single_ URDF file with the layout, because the URDFs can reference other files for meshes and such.

defunctzombie, 2022-04-01:
I think this needs to tie into the "asset manager" concept we've talked about in the past. These assets would also be something we could sync as part of the team account.

jhurliman, 2022-10-06:
Closing, there are no plans to serialize URDFs into layouts

closed 2022-10-06 by jhurliman