foxglove/studio #3123: Add NavSatFix as Displayable topic in 3d Panel

drldcsta, 2022-03-31:
**Current Options**

Currently, the map is only available for display on the map panel, which makes it difficult to tell where exactly the robot is.

**Proposal**
We propose having the NavSatFix as a displayable topic in the 3d panel, and this would render map tiles when toggled on. In addition to this, we think it would be beneficial to switch to Mapbox for map tiles, as opposed to the current OpenStreetMap solution

defunctzombie, 2022-04-01:
To display a message in the 3d panel, we need to know the `frame` of the message and also have a [FrameTransform](https://foxglove.dev/docs/studio/messages/frame-transform) between the `frame` and whatever other objects (and their frames) are in the same scene.

How would we go about doing that for a NavSatFix or [LocationFix](https://foxglove.dev/docs/studio/messages/location-fix) message?

AndrewNolte, 2022-04-04:
> To display a message in the 3d panel, we need to know the `frame` of the message and also have a [FrameTransform](https://foxglove.dev/docs/studio/messages/frame-transform) between the `frame` and [...]

[NavSatFix](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatFix.html) has a header, which has a frame_id, so this could pin down the location. What could be tricky here is aligning the heading/orientation correctly, which isn't in this message. Maybe this could be encoded somehow in the global tf that `frame_id` has.

defunctzombie, 2022-04-04:
> [NavSatFix](https://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatFix.html) has a header, which has a frame_id, so this could pin down the location.

The transforms have information about how one frame relates to another (usually in units like meters and radians). How would you then tie that to the lat/long of the NavSatFix?

The issue I'm stuck on:

I've got a 3d panel visualizing in `base_link`. I receive a NavSatFix message with frame_id `gps_sensor`. If I have a transform from `gps_sensor` to `base_link`, that tells me the location of `gps_sensor` in `base_link`. How would I figure out the location of lat/long from that message in terms of (meters and radians) relative to `gps_sensor` tho?

OgreTransporter, 2022-06-07:
Mapbox I don't find absolutely necessary, but I agree that there should be alternative map sources. The custom layer does not work properly, unfortunately. Also, an offline alternative (see #3511) would be useful.