foxglove/studio #3124: Axis oriented line segment visualization

drldcsta, 2022-03-31:
Currently, our ‘line’ that we use to visualize our path tries to orient itself perpendicular to the camera. I think it would be awesome if foxglove made a new viz type for this that was oriented with the z axis, so that the paths look something like what’s in [this](https://www.youtube.com/watch?v=SbGCIuSyRck&ab_channel=Nuro
) video:

jtbandes, 2022-03-31:
Can you clarify what "our ‘line’ that we use to visualize our path" is? Is there a screenshot or bag that can be shared, or at least a description of how the visualization is set up in the app right now?

And which path are we talking about from the video? This one that shows the car's trajectory? 
[image]

AndrewNolte, 2022-04-04:
> 

Yes, the visualization right now uses a line type (4) with a width set to the width of the rover. Looking on the other viz types, it may be possible actually to achieve what we want with [triangle list](http://wiki.ros.org/rviz/DisplayTypes/Marker#Triangle_List_.28TRIANGLE_LIST.3D11.29_.5B1.1.2B-.5D) However, it would be nice if this transformation was done on the client-side, or within foxglove itself. Perhaps another viz type could be added.

AndrewNolte, 2022-04-04:
An example of the foxglove trying to render the line to be perpendicular to the camera:
[image]
A more dramatic example:
[image]
The data:
[image]

abhilashchowdhary, 2022-04-04:
I believe this is an ask for "first person follow" view  on the selected frame, @AndrewNolte is that correct ?