foxglove/studio #3125: Web Sockets on web client requires wss, not ws, however wss doesn't work

AndrewNolte, 2022-03-31:
**Description**
Trying to connect to a live rover over websockets (ws://<url>:8765) via the web client gives 

> Failed to construct 'WebSocket': An insecure WebSocket connection may not be initiated from a page loaded over HTTPS.

After changing to wss://myurl:8765, I get this non-descriptive error

> Check that the WebSocket server at wss://myurl:8765 is reachable and supports protocol version foxglove.websocket.v1.

This error occurs on the electron app as well. So it may be that the foxglove python server doesn't support wss. wss also doens't work when trying to connect localhost.

- Version: Latest
- OS: Ubuntu 20.04

**Steps To Reproduce**

Try to connect to a non-localhost over foxglove websocket

**Expected Behavior**
It's connected

**Actual Behavior**
It's not connected

defunctzombie, 2022-04-02:
Tragically you are running into different limitations between web and desktop and secure vs un-secure connections.

When using the hosted web version (studio.foxglove.dev), the browser requires that any connection the site establishes to a host be done over a secure connection (https or wss). The browser has an exception for "localhost" which is why connecting to `ws://localhost:8765` would work but not `ws://myhost:8765` as the browser no longer consider the `myhost` connection ok to make from the secure website. Unfortunately, `wss` doesn't work because the websocket server itself (lets say the example python server), is not running with ssl. You could modify it to use ssl but would need to get a certificate which the browser accepts to do that.

For the desktop app there is no "secure" requirement as there is with the hosted browser app. You can open a connection to `ws://myhost:8765` no problem. However, for `wss` the same comments apply; you would need to modify the websocket server to use ssl and get a certificate.

Here's a quick compatibility table I've put together.

| App | Protocol | Host | Status |
| --- | --- | --- | --- |
| web | ws | localhost | ✔ |
| web | wss | localhost |  ✔*  |
| web | ws | myhost | |
| web | wss | myhost |  ✔*  |
| desktop | ws | localhost | ✔ |
| desktop | wss | localhost |  ✔*  |
| desktop | ws | myhost | ✔ |
| desktop | wss | myhost |  ✔*  |

* Requires a server with SSL and valid certificate

/cc @esthersweon Might be worth describing some form of this in our documentation.

I'm going to close the issue given this explanation. If you have any other questions feel free to comment or reach out on slack.

closed 2022-04-02 by defunctzombie

saumilsdk, 2022-10-28:
@defunctzombie i have deployed webserver over cloud with ssl certificate and https endpoint. I need to do ws://172.1.2.3:9090 host running in local network but failing. ws://localhost:9090 works. Please help.