foxglove/studio #3126: Enum state visualization doesn’t work

drldcsta, 2022-03-31:
We would like to be able to plot `state` as seen in the below example however it does not show up 

```syntax = "proto3";
import "src/msgs/header.proto";
 
package pmx;
 
/// An enumerated type to describe the state of the intersection detector
enum IntersectionStateEnum {
 INTERSECTION_STATE_NORMAL = 0;
 INTERSECTION_STATE_WARNING = 1;
 INTERSECTION_STATE_CRITICAL = 2;
 INTERSECTION_STATE_UNKNOWN = 3;
}
 
/// Status message emitted by the intersection detector
message IntersectionState {
 Header header = 1;
 IntersectionStateEnum state = 2;
 float speed_limit = 4;
 
 reserved 3;  // Removed speed_reduction_factor.
}
 
```

jtbandes, 2022-03-31:
Blocked on https://github.com/foxglove/studio/issues/2214

jtbandes, 2022-03-31:
Could we add some more clarification about what "does not show up" means? Are we talking about getting the names "INTERSECTION_STATE_NORMAL" to show up (which is what #2214 is about)?

AndrewNolte, 2022-03-31:
I wont get the names, but it also won't plot the basic number

jtbandes, 2022-04-01:
Cool, turned out to be a simple oversight with a quick fix: https://github.com/foxglove/studio/pull/3130

This fix actually made the enum names show up too, so not actually blocked on #2214 😅

closed 2022-04-01 by jtbandes