foxglove/studio #3127: Lidar points are fixed pixel size, rather than fixed size

drldcsta, 2022-03-31:
I think it would be better if you could zoom in on lidar data and still be able to see the points. I understand that there’s a lo

Lidar data looks great here:
[image]
But not so great here (Zoomed in image so you can sort of see the points)
[image]
[image]

jtbandes, 2022-03-31:
Seems like we would want this to be a setting, it feels unlikely that every user would want their points to zoom with the camera.

amacneil, 2022-04-04:
Could we do something intelligent so that only after a certain zoom level / density the points start getting larger?

AndrewNolte, 2022-04-04:
I think this can be done as a webGL shader. I did something similar or a graphics project where I wanted the viz to look good even when zoomed in. And I think it'll end up looking more natural when zooming in, as the physical size of the points wont change anymore. Right now they are changing, as the point size is defined in relation to the screen, not the 3d environment