foxglove/studio #3128: Better query syntax for proto maps

drldcsta, 2022-03-31:
We have a topic that looks like this:

```
message MetricWithMeta {
    string name = 1;
    repeated Label labels = 2;
    map<string, double> values = 3;
  double values
    map<string, int32> int_values = 4;
    map<string, string> str_values = 5;
}
```
which makes: 
[image]

And for the plot panel, it produces an array of inconsistently ordered key/value pairs, as seen in the screencap. It would be nice if it were structured as a tree, so we could do /metrics/camera_front.metrics_coll.metrics[0].labels.average

jtbandes, 2022-03-31:
Just noting that this should be possible to access the data today, using something like `/metrics/camera_front.metrics_coll.metrics[0].labels[:]{name=="average"}.value`. That's definitely not a great syntax though.

This issue is probably blocked on or at least related to https://github.com/foxglove/studio/pull/3091

AndrewNolte, 2022-04-04:
> Just noting that this should be possible to access the data today, using something like `/metrics/camera_front.metrics_coll.metrics[0].labels[:]{name=="average"}.value`. That's definitely not a great syntax though.
> 
> This issue is probably blocked on or at least related to #3091

Yeah I was able to get it using that syntax, but thought our devs would like it as a tree. I think it that wouldn't be too bad if the plot panel had that as a suggestion, I know it suggests that syntax when the string is a frame_id