foxglove/studio #3129: Add query param to force "open in desktop" or "open in web"

defunctzombie, 2022-03-31:
Sometimes users want to make links that force the desktop app to open. An example would be to open a websocket connection to a non-localhost domain (which won't work with https pages in the browser).

https://studio.foxglove.dev/?ds=foxglove-websocket&ds.url=ws://myurl:8765

Having a query argument that can bypass the launch preference and open the desktop app would allow for sharing such links.

amacneil, 2022-04-04:
Have users actually asked for this feature? If so, it would be better for them to open the issue and share their thoughts.

In this specific case, it seems like creating a desktop link is only treating the symptoms. A better place to fix this would be if we detect a web user tried to connect to an unsupported combination (e.g. insecure non-localhost websocket) we should always show them an explanation and offer for them to open it in desktop (instead of unhelpful error messages as in https://github.com/foxglove/studio/issues/3125).

defunctzombie, 2022-04-07:
> Have users actually asked for this feature?

Yes.

It's trickier to go back to the option screen once this happens. Maybe I want to create links that always open the desktop app if I know they aren't going to work ever in web (like hostnames for websocket connections). This would allow me to do that rather than sending users through a set of dialogs only to then open in desktop.

siddancha, 2022-08-05:
My use case is to force the web app to open, since (1) I know that my users don't use the desktop app and (2) I want to avoid making them go through dialog boxes before reaching the visualization.

I'm not saying we should get rid of the dialog box; I think it is a good default. It would be great if I had the **option** to avoid the dialog box.

amacneil, 2022-08-06:
Makes sense. I could see it being an additional query param, which would bypass the selection screen.

Note that if you force "desktop" and they don't have desktop installed, they will hit our "we tried to launch your desktop app but click here if you want web anyway", because from the browser we have no way of knowing whether the desktop app launch was successful.

camchambers, 2022-08-11:
We host the web app on our robots and only use the web version. Since different robots have different hostnames we get a prompt each time we access a robot.

vbaier, 2022-08-31:
This has actually proven a big issue for prototyping using foxglove in our internal systems to the point where we are considering using webviz to avoid the extra user friction. Our case is simply to launch a visualizer from existing resources that we have already inventoried and this is the only part of the system that does not include low-friction default behavior.

jtbandes, 2022-09-01:
With https://github.com/foxglove/studio/pull/4325 we have changed the default so that links will open in the web app without a prompt. If you still need an override, you can set the `openIn` url param.

PR #4325 (2022-09-01): add url param &openIn=web/desktop to override launch preference
**User-Facing Changes**
Added a URL param `&openIn=[web/desktop]` to override the default behavior of asking the user whether to open urls in the desktop app.

**Description**
Resolves #3129
commit 15399e34 (2022-09-01): add url param &openIn=web/desktop to override launch preference (#4325)

**User-Facing Changes**
Added a URL param `&openIn=[web/desktop]` to override the default behavior of asking the user whether to open urls in the desktop app.

**Description**
Resolves #3129
closed 2022-09-01 by jtbandes