foxglove/studio #3134: ROS parameter fetch failed

Magnusgaertner, 2022-04-01:
**Description**

[image]

Logs: 
```
Error: Expected a double but got 'inf'
    at i._endDouble (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:211:13794)
    at v._onClosetag (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:211:12738)
    at v.u.emit (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:910:5948)
    at u.J._parser.<computed>.d.indexOf.J._parser.<computed> [as onclosetag] (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1062:44912)
    at j (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1063:3620)
    at V (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1063:3659)
    at Ce (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1068:1887)
    at u.oe [as write] (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1070:4920)
    at v.write (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1062:44573)
    at v.end (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1062:44670)
    at file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:211:15470
    at new Promise (<anonymous>)
    at i.deserializeMethodResponse (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:211:15188)
    at i.methodCall (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:211:19267)
    at async i.multiMethodCall (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:211:19329)
    at async a._multiMethodCall (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:10:81912)
    at async a.subscribeParams (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:10:80040)
    at async _.subscribeAllParams (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:10:73159)
    at async II._requestTopics (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1286:7681)
```

- Version:
- OS: ubuntu 20.04

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2022-04-01:
@Magnusgaertner could you share some more details about the parameter or steps to reproduce? Was the parameter set with a launch file? What kind of parameter is it and what you'd set the value to? Any additional details to help us recreate this would be helpful.

jtbandes, 2022-04-01:
According to http://xmlrpc.com/spec.md,

> There is no representation for infinity or negative infinity or "not a number". At this time, only decimal point notation is allowed, a plus or a minus, followed by any number of numeric characters, followed by a period and any number of numeric characters. Whitespace is not allowed. The range of allowable values is implementation-dependent, is not specified.

However it may be that the ROS param server is okay with inf/nan values and represents them as `inf`/`nan` (this is the `str()` representation of inf/nan in python).

defunctzombie, 2022-04-04:
John indicates that ros defers to python parameter parsing and not strictly conforming to xmlrpc.

ROS has added a feature that is not in spec by supporting `inf`/`nan`.

We should add support for this handling to our xmlrpc lib.

defunctzombie, 2022-04-07:
@jhurliman assigning to you cause it sounds xmlrpc/ros protocol work related. You can make the call on what behavior we should support here.

Magnusgaertner, 2022-04-11:
Hi all, 

thank you for the fast responses and the public communication on this issue :+1: 

I really appreciated it :ok_hand:

closed 2022-04-19 by jhurliman