foxglove/studio #3146: Keyboard input sporadically disappears

jackiescanlon, 2022-04-05:
**Description**
I've experienced this problem several times now (and my coworker has as well) where the characters we type in a plotting legend show up briefly, but then disappear, and then sometimes come back and sometimes not. It also is usually not all the characters I type - they are sporadic.

- Version: 1.5.1
- OS: Ubuntu 20.04

**Steps To Reproduce**

1. Open a live connection (I used turtlesim running on the same machine as foxglove in this example).
2. Create a plot panel subscribe to several topics (it seems that this bug is easier to reproduce when Foxglove is given a lot to do)
3. Type out some topic names

**Expected Behavior**
Typed words should appear on the screen as typed

**Actual Behavior**
Typed words get jumbled, with characters randomly dropped, or dropped and then re-added later in the string.

For example, I might try to type "hello/test/123/abc". Right after I type the "ab", they go away, so I'm left with "hello/test/123/c".

Or, the missing characters might come back. For example, if I type "this/is/a/test", the "th" in "this" disappears and doesn't come back, but also, "test" disappears, and _does_ come back after a second, leaving us with "is/is/a/test".

I captured a video of those two examples here, although its harder to understand if you don't see where my hands are typing.

https://user-images.githubusercontent.com/43035541/161804578-f8bd82a7-7024-46a0-a09e-2f15352a0ae6.mp4

Sometimes the characters come back later in the string, which is the worst of all. For example, "test/123/abc" might come back as "test/1/ab23c". I wasn't able to get an example of that on camera though.

foxymiles, 2022-04-12:
Investigating. The log shows this warning over again:

https://github.com/foxglove/studio/blob/470fb0ade84b95f4ce513562e74a1a0c6daff456/packages/studio-base/src/components/TimeBasedChart/index.tsx#L175-L181

defunctzombie, 2022-04-14:
My suspicion is this relates to races with state tracking in the input fields in the legend and updating the panel saved state which is then itself fed as state to the input fields.

defunctzombie, 2022-04-22:
> datasets so even a single character change results in a lot of work.

😭 

Classic lagging UI challenge. Do we debounce the path changes? Maybe that would help mitigate this further.

foxymiles, 2022-04-22:
Debouncing might help and also not changing the subscribed paths until a parseable path is entered.

foxymiles, 2022-04-25:
Another way to repro this:

1. Wait until the plot has spooled up data for a few minutes and starts to get slow.
2. Quickly enter a random gibberish message path like "asdfasdfasf"
3. Hit command-A to select it then back space to delete the whole path.

Repeat this quickly and you can see the updates are dropped entirely sometimes.

foxymiles, 2022-04-29:
Fixed by https://github.com/foxglove/studio/pull/3281 https://github.com/foxglove/studio/pull/3290

jackiescanlon, 2022-06-15:
Unfortunately this is still an issue for me and my team. It seems to happen less frequently now, but when Studio has a high load (lots of topics being plotted, for example), we still see the issue. Here's a short clip of it happening on v1.14.2. I type out `/turtle1/pose.x` perfectly, but Studio drops the `se` in pose.

This isn't a huge deal for topics that already have been published, because we can scroll through the suggested topic names instead of typing them out. This is more a problem for us when we are setting up a plot for a topic that hasn't been published yet, which is a frequent occurrence - we set up a plot to view several topics and then trigger an action that causes data to be published for the first time on those topics. Foxglove doesn't suggest the topic names to us, so we have to type them in, causing us to receive this keyboard jitter.

closed 2022-06-17 by foxymiles

foxymiles, 2022-06-17:
@jackiescanlon I've made some more improvements to the plot legend that will be in the next release. Hopefully this will finally fix this problem for you.

foxymiles, 2022-06-23:
@jackiescanlon Did that most recent fix improve the situation for you?

jackiescanlon, 2022-06-24:
I haven't had access to the high load system where I was seeing this occur the most, but in local testing at least, I haven't had the issue again. Thank you so much!!

PR #3629 (2022-06-17): Use local paths state in plot legend to improve interactivity
**User-Facing Changes**
This improves interactivity of the plot legend message path input.

**Description**
Currently we immediately flush every edit to a plot message path input to  the panel config, and this results in studio doing a lot of work including updating subscriptions, rebuilding callbacks etc and re-rendering the entire plot on every keystroke.

The approach here is for the plot legend to maintain its own local copy of the paths and use a debounced flush to push this back to the underlying panel config.

It also adds a  [... cut]