foxglove/studio #3148: Some node names break node playground typechecking

jtbandes, 2022-04-06:
[image]

[image]

When a node name does not start with `/studio_node`, typechecking is broken in the editor. The node still works at runtime though.

defunctzombie, 2022-04-20:
This happens because the node name is also used as the "filepath" in the "virtual" project structure. Files like `types.ts` are hard-coded to have a file name like `/studio_node/types.ts`. When you change the node name, this becomes its new full path and `./types` is no longer resolved because it is no longer at the same "path".

jtbandes, 2022-04-20:
Yes, I think we need to stop using the node name as part of the filepath. I'm not sure if there was any valid reason to do that but it seems unnecessary.

PR #3223 (2022-04-20): Use the basename from node playground scripts in the editor

**User-Facing Changes**
User can name their node playground nodes whatever they want; they no longer need to keep the `/studio_node/` prefix in their name.

**Description**
The editor uses the script filePath to resolve additional imports. When a user renames their node and removes the /studio_node/ prefix, it causes the relative path of the file to change.

To allow users to name their node whatever they want, we use only the basename of the path and manually prefix the /studio_node/ path since this is an internal implementation det [... cut]
commit 9a219226 (2022-04-20): Use the basename from node playground scripts in the editor (#3223)

The editor uses the script filePath to resolve additional imports. When a user renames their node and removes the /studio_node/ prefix, it causes the relative path of the file to change.

To allow users to name their node whatever they want, we use only the basename of the path and manually prefix the /studio_node/ path since this is an internal implementation detail.

This change also removes the /studio_node/ prefix from new node names since it is no longer a requirement to have this prefix.

Fixes: #3148
closed 2022-04-20 by defunctzombie