foxglove/studio #3152: Missing transform error when there is indeed the transform

charlielito, 2022-04-07:
**Description**
I am trying to visualize a costmap topic on foxglove but I get the following error: `Missing transform from frame <odom> to frame <map>`. Nonetheless, there is indeed a transformation between those 2 frames being published. In fact, I am able to the TF tree in foxglove. I don't think it is related to https://github.com/foxglove/studio/issues/2575 because the costmap is generated by the ROS system

[image]

- Version: Foxglove Studio version 1.6.0-dev
- OS: Ubuntu 20.04

What could be happening?

jhurliman, 2022-04-07:
Are you able to capture this behavior in a recording (`.bag`, `.mcap`, `.db3`) that you could share? That makes troubleshooting coordinate frame and timing issues much more straightforward. If I had to make a guess on what is happening just from the information provided here, I would say there is a timestamp issue somewhere. The relevant TF header.stamp is greater than the costmap header.stamp, or too far in the past, or the costmap header.stamp is set to 0, or something along those lines.

This is a common category of visualization failure, so any real-world examples help inform what error messages to surface or possibly better ways to visualize what transforms exist at which time ranges.

defunctzombie, 2022-04-07:
@jhurliman assigning to you for ownership and follow-up once we hear back with more info.

charlielito, 2022-04-07:
You are right @jhurliman, the header is indeed set to 0 for some reason. I'll investigate this. Thanks for the fast response!

closed 2022-04-07 by charlielito