foxglove/studio #3159: TF Sync with rosbridge over high latency poor

iandouglas96, 2022-04-07:
Under high latency network connections, the tf synchronization using rosbridge fails.

- Version: 1.6.0
- OS: Ubuntu 18.04

**Steps To Reproduce**
[image]

Bagfile and video at: https://drive.google.com/drive/folders/1yMsjnMXOcP3ln3T237nC7hrRHpmUo7cg?usp=sharing

Play bagfile bask over rosbridge from a compute with a slow network connection.

**Expected Behavior**
Even if messages are dropped, topics should always be tranformed using the tf transform with the same timestamp as the message.  See the rviz visualization in the linked video, which is correctly displaying the scene.

**Actual Behavior**
Studio seems to use the most recent available transform or perhaps the closest timestamp available in the buffer, instead of waiting for the appropriately timestamped transform to arrive.  This leads to stuttering and misaligned frames as can be seen in the video.

defunctzombie, 2022-04-14:
Note that rviz does not use rosbridge so this could be a rosbridge performance issue. Can you try reproducing this with the ROS1 native connection in the Studio desktop app to see if it's due to rosbridge?

Yes, Studio uses the latest available transform that can service the transform lookup rather than blocking and waiting. This is by design.

defunctzombie, 2022-05-02:
@iandouglas96 I've not heard back. I am going to close this issue because I think it is specific to rosbridge. Rosbridge for ROS1 does have performance limitations since it is written in python and also de-serializes the message data prior to re-serializing for sending. It is on our radar to improve this for rosbridge.

_by design_ Studio does not block on rendering waiting for a specific transform but uses the most currently available transform to render.

closed 2022-05-02 by defunctzombie

iandouglas96, 2022-05-03:
It is rosbridge specific, but it might be helpful to provide an option to wait for a specific transform, or at least wait for a transform more recent than the target time, in order to mitigate these sorts of problems.  I believe rviz does something like this, since visualization will often only update as frequently as a lower frequency transform in the tf tree.