foxglove/studio #3166: Connect scatterplot points with lines

rgov, 2022-04-09:
This screenshot shows a plot of `msg.x[:]` vs `msg.y[:]`.

[image]

It is difficult to see the data points, let alone identify a trend in the data. Connecting the points with lines, as Studio does for plots of scalar values (rather than arrays), would help a lot here.

... Plus then we can abuse the Plot panel to render vector Etch-a-Sketch graphics.

defunctzombie, 2022-04-18:
Do you expect the lines to connect the points in the order the points appear in the dataset (so lines can connect back and forth across the entire display) or do you expect lines to connect points in increasing x order?

rgov, 2022-04-18:
I would plot them in the order they appear in the array. For my use cases they will already be sorted by _x_ ascending.

This creates some flexibility to do interesting plots like UTM easting, northings of a vehicle's path. 

If someone wants sorted values a Node Playground would be pretty easy. Or a checkbox.