foxglove/studio #3168: Plot y-axis is not scaling properly

rgov, 2022-04-10:
The *y* axis of a Plot panel is not scaling to fit the data. Perhaps this is related to plotting an array of values?

- Version: 1.6.0
- OS: macOS

The Plot panel is plotting an array of values from a message, which is displayed in the upper left. You can see that almost all of the values are greater than 1, yet the *y*-axis scale only goes up to 0.98.

[image]

rgov, 2022-04-10:
Fixing the *y* minimum to 0 seemed to make it snap into shape (different data being shown than in previous screenshot).

[image]

foxymiles, 2022-04-12:
I'm able to reproduce this with this node playground script:

```typescript
import { Input, Message } from "./types";

type Output = {
  values: number[];
};

export const inputs = ["/sample1"];

export const output = "/studio_node/output_topic";

export default function node(event: Input<"/sample1">): Output {
  const values = Array(30)
    .fill(undefined)
    .map((v) => 2 * Math.random());
  return {
    values,
  };
}
```

PR #3184 (2022-04-13): Fix chart auto-y scaling
**User-Facing Changes**
This fixes an issue with plot auto-scaling on the Y axis.

**Description**
The issue is that the downsampling logic was also discarding points outside the current Y axis scale but this was preventing the chart from autoscaling as new data arrives.

Fix #3168
closed 2022-04-13 by foxymiles