foxglove/studio #3172: data source info panel not updating over websocket

AndrewNolte, 2022-04-11:
**Description**
Message count and frequency values show "---" when using foxglove websocket

[image]

- Version: Latest webapp
- OS: Ubuntu 20.04

**Steps To Reproduce**
Connect to a foxglove websocket connection
Open a Data Source Info Panel

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2022-04-14:
Also worth noting that these fields won't be available for topics which do not have active subscribers from other panels since studio does not subscribe to all topics.

Today this is by design for live connections.

Changing this over to a feature request for seeing message counts and frequencies for live connections.

AndrewNolte, 2022-04-14:
Actually it makes sense that it shouldn't work for all topics on live connections. I just noticed that it also isn't working on data platform files either.

closed 2022-04-21 by jhurliman