foxglove/studio #3178: Opening a url for data platform does not take me to my requested timestamp

defunctzombie, 2022-04-12:
When I open a data platform url and play the data I see the URL updates the `time` argument. However, when I refresh or share this url the datasource does not open to the requested seek time.

https://studio.foxglove.dev/?ds=foxglove-data-platform&ds.deviceId=dev_Wm1gvryKJmREqnVT&ds.end=2022-04-12T20%3A12%3A49.126000000Z&ds.start=2022-04-12T20%3A07%3A49.272000000Z&layoutId=ea800f2c-f854-49de-8ef1-a2f7e1f31719&time=2022-04-12T20%3A08%3A31.653456688Z

defunctzombie, 2022-04-12:
Seems to be an issue with the experimental data platform player

foxymiles, 2022-04-19:
Looks like the issue here is that we try to seek the time before the player is ready so it just discards the seek request.

https://github.com/foxglove/studio/blob/bcd44725da190e1259afc306562c5fd85a3892b5/packages/studio-base/src/players/IterablePlayer/IterablePlayer.ts#L890-L895

PR #3211 (2022-04-20): Wait until player is ready to seek to url time
**User-Facing Changes**
This fixes seeking and url times for the experimental data platform player.

**Description**
Currently we try to seek to the time specified in the url before the new data platform player is ready and the seek is just discarded. The new logic here waits until the player is ready before seeking. This also requires changing the new player to not report presence of `PRESENT` until it enters the `start-play` state.

In general the order of operations on starting up a new instance of studio is complex. Maybe player presence is to [... cut]
closed 2022-04-20 by foxymiles