foxglove/studio #3179: Opening a URL with a layoutID for a layout I do not have tells me to select a layout

defunctzombie, 2022-04-12:
* Sign in to your foxglove account.
* Open studio.foxglove.dev and let the layouts sync.
* Open https://studio.foxglove.dev/?layoutId=ea800f2c-f854-49de-8ef1-a2f7e1f31719 and the layout should open
* Delete all the localstorage items for cached remote layouts and refresh the page. The layout will not open and you'll see a `Select a layout in the sidebar to get started!` message.

This is a bad experience because when I make a new shared layout and send it to a team member to view they can't open the link.

foxymiles, 2022-04-18:
Is this the same as #3119 ?

jtbandes, 2022-04-18:
Seems like it to me

jtbandes, 2022-04-18:
Duplicate of https://github.com/foxglove/studio/issues/3119

closed 2022-04-18 by jtbandes