foxglove/studio #3182: Studio publisher does not connect to remote subscribers in ROS1

jessicaaustin, 2022-04-13:
Studio publisher does not connect to remote subscribers in ROS1 -- in other words, if the ROS_MASTER_URI is not pointing to localhost, then messages fail to publish.

I've seen this behavior on our actual systems, and in this ticket I reproduce on a dummy setup. Of course it's possible I have some configuration wrong somewhere, but I am able to publish via `rostopic pub` using a remote host for ROS_MASTER_URI.

- Studio Version: 1.7.0
- OS: Ubuntu
- ROS1 live connection

**Steps To Reproduce**

First, verify publishing works when running all on one host (in this example, my host is called `mimir`):

1. Start `roscore`
2. Run `rostopic echo /chatter`
3. Open Studio and connect to ROS1 with `http://localhost:11311`
4. In studio, add a Publish panel and send a message on `/chatter`
5. Verify the message is received by the `rostopic echo` process

[image]

Then, try publishing when Studio and the ros master are on _different_ hosts:

1. On remote host, start `roscore` (in my example the remote host is called `yeti`)
2. On remote host, run `rostopic echo /chatter`
3. Open Studio and connect to ROS1 with `http://yeti:11311`
4. In studio, add a Publish panel and send a message on `/chatter`

**Expected Behavior**

* The message is received by the `rostopic echo` process on the remote host

**Actual Behavior**

* No message is received
* 
[image]

I can't find an explicit error message in any logs anywhere. I've also triple-checked all of my configuration so I don't think it's that, though I'd love to be proven wrong!

For example, I can use `rostopic pub` to send messages from my local to remote host:

[image]

foxymiles, 2022-04-19:
Thanks for the careful & extensive documentation of the problem as usual!

jtbandes, 2022-04-19:
I'm having some trouble reproducing this problem. It does happen sometimes, but it seems really sporadic and I can't figure out what exactly triggers it.

My setup is Studio running on macOS, and ROS 1 Noetic running in an Ubuntu 20.04 VM. I use the VM's IP address for the ROS_MASTER_URI and the default `.local` domain for ROS_HOSTNAME.

Usually it works fine on the first launch and connection. I've tried various combinations of restarting roscore, restarting rostopic echo, restarting Studio, disconnecting/reconnecting Studio to the ROS master, and changing the advertised topic/datatype in Studio. I would say it works about 80-90% of the time. The other ~10% of the time, Studio successfully invokes the registerPublisher method and hears about existing subscribers from the master, but the rostopic echo subscriber does not initiate a connection to the new Studio publisher.

I'm able to get some debug logs from the ROS master using `tail ~/.ros/log/latest/master.log`, but I don't see any similar debug logs being produced by rostopic echo, which makes it hard to tell why it's not connecting to Studio.

Do you have any more insight on how to get the problem to happen consistently?

jtbandes, 2022-04-19:
Here's a simple subscriber script that produces some debug logs about connections. However, when I use this script, I can't get the issue to reproduce at all. Maybe it's a problem specific to rostopic echo? Have you observed the issue with any other subscribers?

```py
import rospy
from std_msgs.msg import String

rospy.init_node('testsub', log_level=rospy.DEBUG, anonymous=True)

def cb(msg):
    print('Got', msg)

rospy.Subscriber('/chatter', String, cb)
rospy.spin()
```

defunctzombie, 2022-04-20:
> The other ~10% of the time, Studio successfully invokes the registerPublisher method and hears about existing subscribers from the master, but the rostopic echo subscriber does not initiate a connection to the new Studio publisher.

When you experience this - does clicking the "publish" button more times do anything? My initial thought was that maybe the echo subscriber does not subscribe in time for the first publish.

My next thought here is that maybe our ros1 inbound socket handling _glitches_ out on the connection request from the echo subscriber. It might be worth instrumenting the ros1 "node" and our electron-socket use to see if there is an attempted connection from the echo subscriber at all and we just stall in processing it.

jessicaaustin, 2022-04-22:
https://github.com/foxglove/studio/issues/3096 is similar in the sense that a subscriber isn't getting messages from Studio. But in that case, the issue is it's a C++ subscriber, in this case the issue is _something_ about networking.

In that issue, @jhurliman described [here](https://github.com/foxglove/studio/issues/3096#issuecomment-1103070902) how I can run a test using the https://github.com/foxglove/ros1 repo directly, to cut out Studio from the equation and simplify testing. I then took it one step further and added an example at https://github.com/jessicaaustin/ros_demo/tree/main/cpp_subscriber and replicate using docker only, to eliminate possible issues with my local setup. So next I'll try to do the same thing for this issue, to make sure we're all using the same ROS/linux setup to test with.

And if I can't replicate this issue using the ros1 chatter test directly, but I can with Studio, that's a useful data point in itself.

jessicaaustin, 2022-04-22:
I should also say, we are seeing this in our "real" setup pretty consistently, the dummy example I give in the description is to replicate using the most bare-bones setup possible.

jessicaaustin, 2022-04-25:
Ok i was able to demonstrate the same issue using just the foxglove/ros1 library, which (hopefully) simplifies things.

And I set up a demo at https://github.com/jessicaaustin/ros_demo/tree/main/remote_subscriber so that you can run the test via Docker containers, so that eliminates possible differences with our ROS/linux setup.

Unfortunately there still could be differences in networking between the local and remote host, have to think more about how to make that consistent...

So here's what I'm doing now to replicate this issue:

1. On the remote host, follow instructions in https://github.com/jessicaaustin/ros_demo/tree/main/remote_subscriber README to run roscore and a python listener via docker
2. On local host, run the ros1-chatter demo from the ros1 repo
    ```
    git clone the https://github.com/foxglove/ros1 repo   # i'm on aa42ef5e12eb42159627cafcea5c2a91d7a7fc9c now
    yarn && yarn build
    cd examples/ros1-chatter
    yarn
    export ROS_MASTER_URI=http://<remote hostname>:11311
    yarn start
    ```

Here's an example where my local host is called `mimir` and my remote host is called `yeti`:

[image]

Note how the publisher is registered but no messages are coming through.

If in step (1) I run the same docker containers on the local host, the test works fine:

[image]

Curious if you are able to replicate with this setup now!

defunctzombie, 2022-05-03:
@jtbandes I'm gonna pick this up since I think its related to the other issue and I've got some cycles to put towards this.

defunctzombie, 2022-05-03:
@jessicaaustin I ran the instructions from https://github.com/foxglove/studio/issues/3182#issuecomment-1108981631 using @foxglove/ros1 @ d08dfb9d02a644aff93450100ae5fa3a678d840b

I was not able to reproduce the error you are seeing. My chatter_listener (in my VM) was able to connect to the @foxglove/ros1 publisher from my host:

[image]

On my host I did the following in the ros1-chatter example directory of the @foxglove/ros1 repo.

```
export ROS_MASTER_URI=http://10.211.55.6:11311
yarn start
```

One thing I noticed from your last screenshot is that `yeti` looks like it is trying to connect to `127.0.1.1:43291` to subscribe to the chatter publisher when it should be trying to connect to `mimir` or the IP for `mimir`. This tells me that the ros1-chatter publisher is not reporting the correct IP or hostname to roscore when it informs it that it is publishing the topic.

[image]

This is usually resolved by setting the `ROS_IP` or `ROS_HOSTNAME` env var prior to launching the publisher. So in your case you would do the following to launch the chatter publisher.

```
export ROS_MASTER_URI=http://yeti:11311
export ROS_HOSTNAME=mimir
yarn start
```

Similarly, when you open a connection to a ROS1 robot in Studio, there's a field for the ROS_HOSTNAME:

[image]

This is the hostname that Studio will report to roscore as the name that subscribers should use when trying to establish a connection to the publisher. If this name is not resolvable by subscribers that connection won't work.

From reading this thread its not clear to me if you've tried setting the ROS_HOSTNAME environment variable either in Studio during connection or with the CLI publisher. That would be the first thing I would try when using ROS across hosts.

jessicaaustin, 2022-05-03:
@defunctzombie Thanks for taking a look! Yep makes sense about checking ROS_HOSTNAME etc, I tried to make sure I had that all set up properly but of course it's easy to make mistakes. I'll try again today/tomorrow with fresh eyes.

defunctzombie, 2022-05-13:
@jessicaaustin We've fixed some more c++ peering issues in 1.10.1. Please give that a try and see if you have more success with connectivity.

jessicaaustin, 2022-05-16:
@defunctzombie Still seeing the issues with the latest version of studio (1.11.0). But I think I've found something, see below.

I double-checked ROS_MASTER_URI and ROS_HOSTNAME on both sides. 
For example, I have a single terminal with all these env vars, and I either run:

    examples/ros1-chatter$ yarn start  # to publish via foxglove/ros1

or

    rostopic pub /chatter std_msgs/String "data: 'hi from rostopic pub'"   # to publish via python

In the `rostopic pub` case, my messages are getting through to the remote host, and in the foxglove/ros1 case they're not. 
So I think that eliminates possible issues with the env vars, and it's still not working.

I used `netstat` to inspect the processes on each side, to see what sockets were connected in each case. That was interesting...

For the python publisher:
```
# LOCAL (python publisher)
tcp        0      0 0.0.0.0:42589           0.0.0.0:*               LISTEN      8192/python2        
tcp        0      0 0.0.0.0:44981           0.0.0.0:*               LISTEN      8192/python2        
tcp        0      0 <MIMIR_IP>:42589     <REMOTE_IP>:45524      ESTABLISHED 8192/python2        

# REMOTE (subscriber)
tcp        0      0 0.0.0.0:46633           0.0.0.0:*               LISTEN      14604/python3       
tcp        0      0 0.0.0.0:35669           0.0.0.0:*               LISTEN      14604/python3       
tcp        0      0 <REMOTE_IP>:45524      <MIMIR_IP>:42589     ESTABLISHED 14604/python3       
tcp        0      0 127.0.1.1:46633         127.0.0.1:48016         ESTABLISHED 14604/python3    

# subscriber logs
[DEBUG] [1652671489.595677]: connecting to mimir 42589   
[WARN] [1652671489.642622]: data: "hi from rostopic pub"
```

As you can see, there is an ESTABLISHED connection across the hosts, which makes sense since I can see messages coming through.

For the foxglove/ros1 publisher:
```
# LOCAL (ros1 publisher)
tcp        0      0 127.0.1.1:39011         0.0.0.0:*               LISTEN      8280/node           
tcp6       0      0 :::42253                :::*                    LISTEN      8280/node           

# REMOTE (subscriber)
tcp        0      0 0.0.0.0:46633           0.0.0.0:*               LISTEN      14604/python3       
tcp        0      0 0.0.0.0:35669           0.0.0.0:*               LISTEN      14604/python3       
tcp        0      0 127.0.1.1:46633         127.0.0.1:48016         ESTABLISHED 14604/python3       

# subscriber logs
[DEBUG] [1652671568.776789]: connecting to 127.0.1.1 39011
```

The `connecting to 127.0.1.1 39011` jumped out at me -- the foxglove/ros1 publisher is listening on `127.0.1.1:39011`, _not_ `0.0.0.0:39011` like you would expect.

I haven't had a chance to dig into the code yet to see where this might be happening, but I'll try tomorrow. And maybe it will be obvious to you guys anyway.

Thanks for reading and sticking with me on this one!

defunctzombie, 2022-05-16:
@jessicaaustin That's an interesting find - thank you for doing the deep drive with netstat.

The place where Studio desktop "listens" is in the electron-socket library here:

https://github.com/foxglove/electron-socket/blob/main/src/preloader/TcpServerElectron.ts#L20

We don't actually specify a network interface but rather pass along the _hostname_. What might be happening is the hostname from Studio (specified in ROS_HOSTNAME) is resolving to 127.0.0.1. could be a manual entry in `/etc/hosts` or some other configuration. Either way, the end result is that the _listen_ call seems to think it should only listen on 127.0.0.1 rather than 0.0.0.0 based on what you are seeing above.

What happens if you use the IP address of your host (that the remote host can resolve) in ROS_HOSTNAME within Studio rather than a human hostname? I wonder if that exhibits the same behavior with listen.

jessicaaustin, 2022-05-17:
@defunctzombie Yep that does it! 

Previously, I was setting ROS_HOSTNAME to my machine hostname (`mimir`), and if I set it to the IP address on my VPN that the remote host can actually connect to, then it works both in the ros1 chatter example, and via Studio as well. I just tried in our actual setup, and it works there too. 

So this is great to have a workaround for this issue. Do you think there needs to be anything changed in the foxglove/ros1 library? Based on how other ROS tools work, I'd expect if I set ROS_HOSTNAME to a hostname, and then make sure all relevant hosts can resolve that to an IP address, then I should be good to go.

defunctzombie, 2022-06-13:
> Do you think there needs to be anything changed in the foxglove/ros1 library?

I suspect that might be where any such change needs to happen - based on the above I'm thinking that we should have the node listen on 0.0.0.0 rather than the hostname but maybe that will result in other undesirable properties.

An instructive data point would be to identify what the ros1 c++ and python libraries are doing. Are they always listening on 0.0.0.0 or doing something else with the ROS_HOSTNAME to end up listening on the correct IP.

Another approach would be to allow you to pick the network interface to listen on when establishing the connection in Studio - but I suspect that might end up being more confusing for most users

PR #4400 (2022-09-12): Listen on all interfaces for Ros1Player

**User-Facing Changes**
When publishing messages from studio using the ROS1 native connection, the internal ROS node within Studio will listen on all network interfaces.

**Description**
This change updates the Ros1Player to listen on all interfaces for the rosnode internal to studio. This mirrors the ros_comm c++ behavior.

Fixes: #3182
commit 05e8c760 (2022-09-12): Listen on all interfaces for Ros1Player (#4400)

This change updates the Ros1Player to listen on all interfaces for the
rosnode internal to studio. This mirrors the ros_comm c++ behavior.

Fixes: #3182
commit 071c8b3a (2022-09-15): WIP feature(StreamByImportId): stream by importId + start/end date

WIP feature(StreamByImportId): coverage-by-import and topics-by-import are happening

enhancement(StreamByImport): uses the new topics and coverage endpoints

3D beta rendering support for SceneEntity `texts` (#4387)

**User-Facing Changes**
 - Allow for the rendering of the Text scene entity primitive

**Description**
- added textprimitive class to scene entities using the LabelPool from
the renderer
  - updated scene entity story to include text example

Fixes #4319

Add geojson message to correct feature group (#4399)

The  [... cut]
closed 2022-09-12 by defunctzombie