foxglove/studio #3188: URL's layoutId does not update correctly when there is no data loaded

esthersweon, 2022-04-13:
**Description**
- URL's layoutId does not update correctly when there is no data loaded

- Version: 1.7
- OS: Repro'd on web and macOS

**Steps To Reproduce**

- Open Studio app (web or desktop) without any data loaded
- Click on any layout in the sidebar

**Expected Behavior**
- URL updates to have a `layoutId` param

**Actual Behavior**
- URL does not update to have a `layoutId` param

foxymiles, 2022-04-14:
In fixing this we should be careful not to undo the fix in https://github.com/foxglove/studio/pull/3031

defunctzombie, 2022-04-14:
@foxymiles How do we make this url state stuff more robust? It seems we are constantly thrashing previous fixes.

foxymiles, 2022-04-14:
> @foxymiles How do we make this url state stuff more robust? It seems we are constantly thrashing previous fixes.

It's tough IMO since we're compressing a lot of app state and behavior into a small state vector. Our app state is spread over several independent context and atomic state changes and updates across all of them is very challenging.

PR #3278 (2022-04-28): Make URL state tracking more robust
**User-Facing Changes**
The `layoutId` url parameter (web app) updates when changing layouts and no data source is selected.

**Description**
This fixes an issue with layoutId not being tracked in the absence of a datasource. Currently `PlayerManager` has a separate notion of `selectedSource` and the actually active player. This makes managing URL state very difficult because these two things can be out of sync. This patch removes `selectedSource` as a separate state variable from `PlayerManager` and moves the `sourceId` that the url state hook needs into th [... cut]
closed 2022-04-28 by foxymiles