foxglove/studio #3192: Teleop panel tells me to change settings but the settings menu item is disabled

defunctzombie, 2022-04-15:
When I open the telop panel and connect to a websocket data source, the panel tells me to select a topic in the panel settings. Tragically when I click the cog icon in the panel toolbar and try to open the settings the menu item is disabled.

[image]

---

Related - maybe we should show a cog icon in the text so we teach the user where to find the settings.

foxymiles, 2022-04-15:
I think we're going to have to expose a method in the extension panel API to open the settings panel programatically.

defunctzombie, 2022-04-15:
> I think we're going to have to expose a method in the extension panel API to open the settings panel programatically.

In this particular case the panel toolbar is responsible for the context menu which is outside the extension api. Somehow it has decided this item should be disabled.

foxymiles, 2022-04-15:
I think it's because the Telop panel doesn't expose our older-style config schema. We'll need to change this logic.

https://github.com/foxglove/studio/blob/028a902c1a9aad103066597d97268d93beca0286/packages/studio-base/src/components/Panel.tsx#L592-L604

foxymiles, 2022-04-18:
@defunctzombie @2metres Seems like it would make sense to just always enable this since most panels are going to have something in the settings panel and in the rare case we can just show a "no settings available" message. Otherwise we'll have to make it a subscriber to the panel settings tree and wait until the panel actually pushes a settings tree to know if we should show it or not.

defunctzombie, 2022-04-18:
> @defunctzombie @2metres Seems like it would make sense to just always enable this since most panels are going to have something in the settings panel and in the rare case we can just show a "no settings available" message. Otherwise we'll have to make this component a subscriber to the panel settings tree and wait until the panel actually pushes settings to know if we should show it or not.

Always enable for now sounds good. Once we transition to the new panel toolbar UI proposed by @2metres we can revisit refining this behavior for panels without settings.

PR #3210 (2022-04-19): Always show panel settings menu item in panel actions dropdown
**User-Facing Changes**
This fixes an issue with the `panel settings` menu item in the panel actions dropdown.

**Description**
Currently we disable this menu item if we think the panel has no settings to offer. We don't offer any mechanism for extension panels to control this though and since we expect most panels will have some kind of settings to expose in the panel settings panel it makes sense to just enable this by default and have a a reasonable "no settings" state for the settings sidebar.

This also fixes an issue with cle [... cut]
closed 2022-04-19 by foxymiles