foxglove/studio #3193: Teleop panel settings feedback

defunctzombie, 2022-04-15:
Some feedback from using the new teleop settings (shown below):

[image]

- Remove `See docs here` text. This space should go towards showing the settings rather than space for a link that power users won't click.
- `Filter by layer name` needs a better name. Have we determined what the `layer` concept means here? I do not think `layer` is the correct name. Maybe `group` or just call it `Filter`.
- `Settings` should be called `General`. The entire sidebar is already settings.
- The fields and controls all feel too sparse. Should they be more dense?
- Import/Export button and reset buttons feel too prominent for their value. Maybe they should be side-by-side and not try to fill the entire horizontal space. Maybe they should be text buttons. Maybe there should be a "three dot" context menu next to `Panel Settings` header text and they should be behind that.

Related: https://github.com/foxglove/studio/issues/3053

foxymiles, 2022-04-15:
Seems like in this case the layer filter is just a distraction and we should disable it.

foxymiles, 2022-04-20:
The top label (currently `Settings`) for teleop can be controlled per-panel by setting the root level `label`.

defunctzombie, 2022-05-14:
My main concerns from this have been addressed. The experience is better. I still think there's room to improve the density of the sidebar but that is lower priority.

closed 2022-05-14 by defunctzombie