foxglove/studio #3198: Indicate a loading error to the user when we can't load their mesh resource

defunctzombie, 2022-04-16:
**Description**
When a user provides a robot description to the 3d panel that contains a mesh type that we do not yet support (i.e. obj), we should show them an error indicating we failed to load their URDF properly rather than silently do nothing.

Try loading the URDF from as the robot model in the 3d panel:

```
https://github.com/szenergy/nissan_leaf_ros
package://nissan_bringup/param/nissan_leaf_robotdescription.ref.xml
```

Notice how nothing loads and there are no errors.

Related: https://github.com/foxglove/studio/issues/3088

defunctzombie, 2022-04-16:
@jhurliman Assigning to you since this is 3d panel related. I consider it a bug that we fail silently. We don't need to implement `.obj` loading in the existing panel but we do need to avoid failing silently.

defunctzombie, 2022-04-28:
@jhurliman FYI this issue is about the current 3d panel rather than the new 3d panel. We need to fix this in the current one. I imagine the fix here is to inform the user of any attempt to load an unsupported extension rather than failing or ignoring silently.

jhurliman, 2022-06-27:
New3D now displays topic errors for failed mesh resource loading, and also supports the OBJ format.