foxglove/studio #3199: Studio crashes when running Turtle Sim

WaelKarkoub, 2022-04-17:
I was running it through ROS2 native, on the desktop app on ubuntu 20.04. I tried visualizing `/cmd_vel` topic from turtle sim and it keeps crashing. `Cannot use 'in' operator to search for 'toJSON' in undefined`

```
Error stack:

file:///opt/Foxglove%20Studio/resources/app.asar/renderer/8871.js:646:38868
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1364:11505
l (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1699:11472)
b (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1699:12886)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:72288
xh (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1002:1485)
mm (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1004:5445)
C.unstable_runWithPriority (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1078:3956)
Ms (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1000:39733)
Wf (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1004:1905)
z0 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1002:11651)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1000:39959
C.unstable_runWithPriority (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1078:3956)
Ms (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1000:39733)
Wd (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1000:39903)
Rc (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1000:39838)
El (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1002:8853)
cp (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1000:60927)
listener (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1601:12285)
lv._listener (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1348:5470)
lv._emitState (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:13502)
lv._onPlayerState (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:13251)
async oO._listener (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:13581)
Component stack:

c (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:71972)
re (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1601:9644)
rT (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:14373)
$3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:91859)
K3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:91228)
X3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:86339)
u (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1699:9343)
c (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:71972)
g (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1699:8437)
Ph (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1269:4981)
u (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1699:4852)
E3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:77896)
c_ (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:106041)
b3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:77376)
eI (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:114956)
W (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1039:35352)
aT (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:19120)
W3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:85456)
L3 (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:83190)
Ys (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1269:47882)
p (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1378:303)
div
bS (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1309:45883)
Wn (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1082:6268)
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:65759
file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:3627
we (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:4200)
me (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:3943)
Se (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:4320)
Ua (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1797:65825)
dS (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1309:42901)
aI (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1349:118168)
eP (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1364:3029)
```

jtbandes, 2022-04-18:
Which version/distribution of ROS 2 are you using? (Foxy, Galactic etc)

foxymiles, 2022-04-18:
@jtbandes I can repro this on Foxy.

jtbandes, 2022-04-18:
Cool – so that probably means the root cause is somewhere in the ros2/cdr parsing layer?

foxymiles, 2022-04-18:
Seems likely. You can see there is data in the message:

[image]

closed 2022-04-27 by jhurliman