foxglove/studio #3203: Invalid WebSocket URL pops up toast behind the open dialog

jtbandes, 2022-04-18:
When entering an invalid url such as `test` or `test:` in the websocket url field, the error pops up behind the open dialog in a toast notification. It would be better to display any error inline in the dialog instead.

[image]

defunctzombie, 2022-04-20:
Picking up this issue as it has definitely annoyed me several times and I'd say is pretty terrible ux for what should otherwise be a simple text input error.

jtbandes, 2022-04-20:
Additionally, the user should not be required to enter `ws://`, we should assume `ws://` if no scheme is specified.

PR #3224 (2022-04-22): Add a validate property to IDataSourceFactory formConfig fields

**User-Facing Changes**

Rather than receive errors for an invalid url format after clicking "open" in the connection dialog, the user is shown an error under the text field and prevented from clicking "open" until they resolve the issue.

[image]
[image]
[image]

**Description**
A data source factory can provide a validate function for each field. The validate function allows the factory to intercept new values and return an error if the value for the field is not valid. The open dialog uses this to determine whether to show an  [... cut]
commit c8b2533d (2022-04-22): Add a validate property to IDataSourceFactory formConfig fields (#3224)

A data source factory can provide a validate function for each field. The validate function allows the factory to intercept new values and return an error if the value for the field is not valid. The open dialog uses this to determine whether to show an error for the field or allow the user to click the "open" button.

Fixes: #3203
closed 2022-04-22 by defunctzombie