foxglove/studio #3207: Incompatibility with WSL (Windows Subsystem for Linux)

robokamran, 2022-04-19:
**Description**
Foxglove Studio works perfectly fine over both rosbridge and direct connection to ROS1 with WSL2.
But for ROS2, the direct connection doesn't work. In fact, when the WSL is active, it cannot even see other networks and domains either. To make it work, I need to disable `vEthernet (WSL)` in `Control Panel\Network and Internet\Network Connections`.
For ROS2, the rosbridge doesn't work either but that's under investigation in another thread.
Please let me know if anyone has successfully used Foxglove with WSL2 in ROS2.

- Version: Foxglove Studio version 1.7.0
- OS: Windows 10, WSL2

jtbandes, 2022-04-19:
Thanks for the report. Could you share some more detailed steps to reproduce the problem? Does disabling `vEthernet (WSL)` make it work or are you still having trouble after using this workaround?

robokamran, 2022-04-20:
Sure @jtbandes, I can.

My computer is connected through ethernet to a local network. On the local network, there are robots and other computers using ROS2. So everyone should be able to see the topics as long as the **ROS_DOMAIN_ID** is the same.

On my computers (both laptop and desktop), when `vEthernet (WSL)` is enabled, I cannot see the messages/topics on the local network. Yes, disabling `vEthernet (WSL)` will make it work, and I can see the topics again. WSL can be running or shut down; either way, it works (if WSL is running and `vEthernet (WSL)` is disabled, only the internet connection of WSL will be lost).

Aside from that, there is no way that I can see the messages/topics from the nodes running on WSL. So it doesn't matter if `vEthernet (WSL)` is enabled or if the computer is connected to another network.

On another topic, but maybe related, I have a USB-Ethernet dongle through which I connect to a robot directly with static IP (sometimes I even share my PC's internet with the robot by it). Again I can connect to `roscore` on ROS1 and diag the robot, but it doesn't work for ROS2.

Please let me know if more info is needed.

jtbandes, 2022-04-20:
Thanks for providing more details. To clarify, it sounds like you are not running any of the software (ROS or Foxglove Studio) _inside_ of WSL – it's just that you have WSL installed?

I wonder if this issue is related? https://github.com/microsoft/WSL/issues/5671

robokamran, 2022-04-21:
My PC is connected to the local network through ethernet and some robots are on the local network. I connect to other robots via Wi-Fi and a separate USB-Ethernet adapter as well. I also run simulations on multiple ubuntu instances in WSL (noetic, melodic, galactic, and so on). And I use Foxglove Studio on Windows as a diagnostic tool for all the robots inside WSL and other networks. For now I don't use ROS2 on Windows.

For ROS1 both rosbridge and direct connection of Foxglove work fine. The problem is only for ROS2.

I checked #5671 issue; I think that's not the case right now because I can easily broadcast on UDP with this on all WSL instances: `nc -u -b 255.255.255.255 9000`

robokamran, 2022-04-21:
### _UPDATE:_
Maybe the problem is not just with the WSL but happens when having multiple network connections in Windows.
BTW Foxglove Studio is an excellent tool for Windows to visualize and debug a robot.

**Environment**
- Ethernet connection to a local network (with internet access).
- Wi-Fi connection to the robot.
- Foxglove Studio 1.8.0 on Windows.
- OPTIONAL: any other interfaces including WSL `vEthernet (WSL)`, USB-Ethernet dongle `Ethernet 2` etc.

**_Steps to reproduce_**
1- Just be connected to any network (for internet connection).
2- Connect to the Wi-Fi hotspot created by the robot.
3- Run a simple publisher on the robot.
4- Open Foxglove Studio and try to connect on the proper `ROS_DOMAIN_ID`.

**_Expected behavior_**
Foxglove Studio should show the topics and messages.

**_Actual behavior_**
Foxglove Studio shows nothing.

**_Tested Hotfix_**
Disable all other network connections in `Control Panel\Network and Internet\Network Connections` except the one connected to the robot. Then restart Foxglove Studio and connect to the `ROS_DOMAIN_ID`.

jtbandes, 2022-04-21:
It sounds like this is a known limitation with our ROS2 / RTPS implementation, which currently only listens on one network interface: https://github.com/foxglove/rtps/issues/17

robokamran, 2022-04-22:
I see, so if this limitation is attended to in another thread, I guess we can close it here since it's not only related to WSL.
I appreciate your support so far.
I'll add a comment to the other thread for my tested hotfix and propose some suggestions.

defunctzombie, 2022-06-13:
Closing this issue as discussed. We've got a new issue tracking listening on all interfaces for rtps.

closed 2022-06-13 by defunctzombie