foxglove/studio #3208: MarkerArray display error in Vmware Ubuntu20.04

darduck18, 2022-04-19:
**Description**
display example datasource /semantic_map,
I have encountered the same problem as #2727, windows is ok but Ubuntu failed.
And I tried other computers with the same problem.
Can i fix it in virtual machine?

- Version:V1.7.0
- OS:Vmware Ubuntu20.04

**Steps To Reproduce**

**Expected Behavior**
[image]
[image]
[image]

**Actual Behavior**

jtbandes, 2022-04-19:
Hi, thanks for the report. Based on the error message, which comes from [here](https://github.com/foxglove/regl-worldview/blob/92df7aad8d28d15a811d5daaed672261c507e27a/src/commands/Lines.js#L533), this error doesn't seem like it should be platform-specific and looks like more of a logic error. Could you share a video of steps to reproduce the error? I have tried enabling the /semantic_map topic and playing through the whole sample dataset on ubuntu, and I can't reproduce the issue. Does it happen 100% of the time for you?

darduck18, 2022-04-20:
> Hi, thanks for the report. Based on the error message, which comes from [here](https://github.com/foxglove/regl-worldview/blob/92df7aad8d28d15a811d5daaed672261c507e27a/src/commands/Lines.js#L533), [...]

thanks for your reply.
it happen 100% for my vmware workstation,I just enable /semantic_map topic
Others computer's issue as below.

[image]

 as #2727, i think it happen only in virtual machine ubtuntu.

jtbandes, 2022-04-20:
What operating system & architecture is your host machine?

jtbandes, 2022-04-20:
Since we can't reproduce this issue, it would help if you can set a breakpoint where the error occurs using the Chrome debugger, and share a screenshot of the variable values, like this:

[image]

Notice in my screenshot, `numInputPoints` is 2820 while `instances` and `poses` are 2819 (that's expected because 2820 points turn into 2819 line segments).

It would also be helpful to see the state after the call to `groupLinesIntoInstancedLineLists()` inside WorldMarkers:

[image]

This is where the original 102 `lineStrip` markers are grouped into one line with 2820 points. NaN points are used to separate multiple segments:

[image]

jtbandes, 2022-04-21:
If you could get the debug screenshots from https://studio.foxglove.dev instead of the desktop app, it should have better variable names (fixed in https://github.com/foxglove/studio/pull/1880)

darduck18, 2022-04-22:
> If you could get the debug screenshots from https://studio.foxglove.dev instead of the desktop app, it should have better variable names (fixed in #1880)

ok,i got it.
[image]
[image]
[image]

By the way, i check this issue in vm ubuntu16.04,it works fine.

jtbandes, 2022-04-22:
>  [image]

In this screenshot, `instances` is 2819, and `poses.length` is also 2819, so there should be no error. Is there sometimes an error and sometimes no error?

darduck18, 2022-04-26:
> > [image]
> 
> In this screenshot, `instances` is 2819, and `poses.length` is also 2819, so there should be no error. Is there sometimes an error and sometimes no error? Can you move the breakpoint to `console.error` so it only stops when there is an error?

I check many times,there is no error,but it also display green background.I think the issue is no case by this error.

defunctzombie, 2022-05-02:
From my reading of the issue it sounds like the problem originally shown in the console (expected poses count was wrong) is no longer happening so I don't understand why it appeared in the original console logs.

We've not been able to reproduce this issue so our latest thinking is that its specific to your VM or graphics driver. Given the disconnect in our ability to reproduce this I am not sure we're gonna be able to provide a fix. If you are able to grab the source code for Studio and run a development build you might be able to poke around and figure something out. We are actively moving to a different rendering engine for the 3d panel. That work is still a month or more away but it might help in cases like this one.

I'm going to close this issue out since I don't see what else we can troubleshoot here. If you do learn more please comment with a way we can attempt to reproduce.

closed 2022-05-02 by defunctzombie