foxglove/studio #3213: Support robot description for 3d models as a string topic

Aposhian, 2022-04-19:
Rviz supports loading 3d models from robot_description published to a string topic. Foxglove does not support this, which means that Foxglove doesn't have feature parity with Rviz. While it it can be worked around by republishing a mesh as a `visualization_msgs/Marker`, it would be nicer if Foxglove worked out of the box with systems that use `robot_state_publisher`.

jtbandes, 2022-04-19:
Can you share some documentation or an example of how this feature works in rviz?

From [docs](https://wiki.ros.org/robot_state_publisher):

> robot_state_publisher uses the URDF specified by the parameter `robot_description` and the joint positions from the topic `joint_states` to calculate the forward kinematics of the robot and publish the results via tf.

This doesn't mention anything about the description being on a string topic.

https://wiki.ros.org/rviz/DisplayTypes/RobotModel:

> Robot Description: The parameter to retrieve the urdf from. Uses searchParam() to search up the parameter tree for the value specified.

Studio does support automatically loading the `robot_description` param in the URDF Viewer panel. Maybe the ask is to extend this feature to the 3D panel as well?

Aposhian, 2022-04-20:
I should have clarified: this is for Rviz and `robot_state_publisher` in ROS2.
The ROS2 version of `robot_state_publisher` accepts the `robot_description` as a parameter, and then republishes it on a topic as a string. https://github.com/ros/robot_state_publisher#published-topics

In `rviz2`, when you add a "Robot Model" you can choose the description source to be by "Topic" or by "File". If by "Topic" you can specify a topic to load
[image]

jtbandes, 2022-04-20:
I see, I didn't realize that params in ROS 2 are not global. That explains the difference between the rviz2 behavior and the documentation site. Thanks for clarifying.

closed 2022-07-14 by jhurliman