foxglove/studio #3216: Open dialog has two scrollbars

jtbandes, 2022-04-19:
Related to #3209? (cc @foxymiles) Open dialog has two scrollbars when the page is vertically constrained:

https://user-images.githubusercontent.com/14237/164086612-73f2d765-c28a-4da6-b35e-f73d294de2b2.mov

foxymiles, 2022-04-19:
There are a lot of elements in the app that don't behave well when constrained this much vertically or horizontally. Do we plan to fix them all?

jtbandes, 2022-04-19:
That's a fair question. Here is my 5¢:

The app does not have to look _nice_ at small sizes, but it should be usable "within reason". Obviously there are limits, for example if the window is smaller than the playback bar itself, nothing other than the playback bar is going to be usable. But with a few hundred pixels of height, most of the content _can_ be used even if it is awkward and requires scrolling.

When I see two scrollbars like this, it tells me something is wrong with the CSS "architecture" (if you can call it that) because we shouldn't have `overflow: auto` on more than one element in a hierarchy if we don't intend for there to be two layers of scrollable content.

All this really just boils down to an assumption about what the app's "usable min height" is. I guess the current dialog works fine if you assume the min height is ~500. The sidebar buttons and sidebar content work down to a min height of ~200 before the icons start overlapping (which is roughly the actual min height of the window on macOS). Since the app is meant to be flexible and can even be used on some mobile devices, I would tend to try and avoid knowingly adding UI that increases the usable min height.

defunctzombie, 2022-04-20:
/cc @2metres for thoughts on how we can avoid such things via the re-use of well behaving widgets or components

closed 2022-06-23 by jtbandes