foxglove/studio #3227: Add option to copy current timestamp in ISO format

amacneil, 2022-04-21:
**Description**
Creating events in Console requires the timestamp in ISO 8601 format:
[image]

Studio only gives you the timestamp in time-only or unix formats:
[image]

To simplify the process of creating events, we need a way in Studio to copy the current timestamp in the format that Console expects it.

In the future we want to allow directly creating events from Studio, but this could be a good short term solution.

(request from Connor Buckland in community Slack)

cc @defunctzombie for triage

bryfox, 2022-04-26:
Certainly doable; I think the trick will be the zone abbreviations — may need to pull in a tz database/library or find some way to work with that using `Intl`.

amacneil, 2022-04-27:
understood re taking up more space. the goal of this ticket is just making copy pasting easier, we can solve that in the studio side or the console side.

although FYI the default studio format is `8:28:47.746 PM PDT` which doesn't give me the date so it's not going to be helpful. maybe we can make it easier to copy the full ISO8601 somewhere in studio.

defunctzombie, 2022-04-28:
> although FYI the default studio format is `8:28:47.746 PM PDT` which doesn't give me the date so it's not going to be helpful. maybe we can make it easier to copy the full ISO8601 somewhere in studio.

It could be nice to show a tooltip? when you hover over the timestamp that has the timestamp in unix epoch or human readable (ISO). You don't have to change your settings to copy either one.

@2metres thoughts on using a tooltip here or if there's a better control or technique?

amacneil, 2022-04-28:
> @ github. com>

How do you copy the text from the tooltip?

defunctzombie, 2022-04-28:
> How do you copy the text from the tooltip?

Thats the age old question.

:trollface: maybe we need a panel who's sole purpose is to show you the current timestamp in every possible format.

amacneil, 2022-04-28:
Just stick a small copy icon in the text box, and when you click it copy the full timedatestamp rather than the time-only timestamp. No one is going to be confused by that.

defunctzombie, 2022-04-29:
> Just stick a small copy icon in the text box,

We are already constrained on space in that text box at times. I often see the time run past the end of the display. We'll see what we can come up with.

defunctzombie, 2022-06-30:
I'm gonna close this out since we've completed the work to support pasting the timestamps that Studio does provide. If we get more feedback that folks want to see the text in another format we can re-visit the specific ask within Studio.

closed 2022-06-30 by defunctzombie