foxglove/studio #3241: Clicking the "visibility" toggle for the Topic settings group does not hide the topics

defunctzombie, 2022-04-22:
**Description**
On the map panel, when I click the 👁️ icon to toggle visibility on the topics group the topics are still visible.

[image]

[image]

defunctzombie, 2022-04-22:
@foxymiles assigning to you since this is from your recent change to migrate the map panel to the settings sidebar

foxymiles, 2022-04-27:
We've temporarily turned these off until we have thought through a bit more how we want them to work.

closed 2022-04-27 by foxymiles