foxglove/studio #3248: Map panel is not rendering GeoJSON markers correctly

foxymiles, 2022-04-22:
**Description**
The map panel fails to find the leaflet images needed to render GeoJSON parkers.

- Version: 1.8.0
- OS: macOS

**Steps To Reproduce**

Send a GeoJSON message to the map panel. This issue is also visible in the storybok.

**Expected Behavior**
The marker should render.

Possibly related: https://stackoverflow.com/questions/60174040/marker-icon-isnt-showing-in-leaflet

**Actual Behavior**
The marker renders as a broken image and there are errors like this in the console:

```data:image/png;base64,i...)marker-shadow.png net::ERR_INVALID_URL```

[image]

tanpinsiang, 2022-04-25:
I am unable to get multiple GeoJSON features to work on the map panel. Could you share your JSON schema and JSON payload for this example?

foxymiles, 2022-04-25:
@tanpinsiang You can see the GeoJSON we use for this storybook story here:

https://github.com/foxglove/studio/blob/47c358e3f4c8e8f1a5bd1188b1b1acc2659948df/packages/studio-base/src/panels/Map/index.stories.tsx#L256-L296

foxymiles, 2022-04-25:
@tanpinsiang More specifically the actual GeoJSON message here:

https://github.com/foxglove/studio/blob/47c358e3f4c8e8f1a5bd1188b1b1acc2659948df/packages/studio-base/src/panels/Map/index.stories.tsx#L256-L296

closed 2022-04-25 by foxymiles

tanpinsiang, 2022-04-26:
Thanks for the help, I am using the ws protocol foxglove.GeoJSON. I tried the GeoJSON above, however, the map panel is blank and this is the error from chrome console. 
My platform is Ubuntu 20.04 Chorme Version 100.0.4896.127 (Official Build) (64-bit)
[image]

`react-dom.production.min.js:216 TypeError: Cannot read properties of undefined (reading 'lat')
    at LatLngBounds.contains (leaflet-src.js:1263:17)
    at FilteredPointLayer (FilteredPointLayer.ts:28:26)
    at MapPanel.tsx:309:37
    at fk (react-dom.production.min.js:262:359)
    at exports.unstable_runWithPriority (scheduler.production.min.js:18:343)
    at gg (react-dom.production.min.js:122:325)
    at Oj (react-dom.production.min.js:261:308)
    at react-dom.production.min.js:261:215
    at V (scheduler.production.min.js:16:224)
    at MessagePort.F.port1.onmessage (scheduler.production.min.js:12:346)
Ni @ react-dom.production.min.js:216
Pi.c.callback @ react-dom.production.min.js:216
Eg @ react-dom.production.min.js:131
Yi @ react-dom.production.min.js:220
dk @ react-dom.production.min.js:259
exports.unstable_runWithPriority @ scheduler.production.min.js:18
gg @ react-dom.production.min.js:122
Uj @ react-dom.production.min.js:252
Lj @ react-dom.production.min.js:243
(anonymous) @ react-dom.production.min.js:123
exports.unstable_runWithPriority @ scheduler.production.min.js:18
gg @ react-dom.production.min.js:122
jg @ react-dom.production.min.js:123
ig @ react-dom.production.min.js:122
fk @ react-dom.production.min.js:263
exports.unstable_runWithPriority @ scheduler.production.min.js:18
gg @ react-dom.production.min.js:122
Oj @ react-dom.production.min.js:261
(anonymous) @ react-dom.production.min.js:261
V @ scheduler.production.min.js:16
F.port1.onmessage @ scheduler.production.min.js:12
index.tsx:16 Uncaught TypeError: Cannot read properties of undefined (reading 'lat') https://studio.foxglove.dev/main.8394f81880db3b03facf.js 117344 15 TypeError: Cannot read properties of undefined (reading 'lat')
    at LatLngBounds.contains (leaflet-src.js:1263:17)
    at FilteredPointLayer (FilteredPointLayer.ts:28:26)
    at MapPanel.tsx:309:37
    at fk (react-dom.production.min.js:262:359)
    at exports.unstable_runWithPriority (scheduler.production.min.js:18:343)
    at gg (react-dom.production.min.js:122:325)
    at Oj (react-dom.production.min.js:261:308)
    at react-dom.production.min.js:261:215
    at V (scheduler.production.min.js:16:224)
    at MessagePort.F.port1.onmessage (scheduler.production.min.js:12:346)
window.onerror @ index.tsx:16
instrument_global.onerror @ instrument.js:506
error (async)
instrumentError @ instrument.js:496
instrument @ instrument.js:45
addInstrumentationHandler @ instrument.js:63
_installGlobalOnErrorHandler @ globalhandlers.js:53
GlobalHandlers2.setupOnce @ globalhandlers.js:39
setupIntegration @ integration.js:44
(anonymous) @ integration.js:58
setupIntegrations @ integration.js:56
BaseClient2.setupIntegrations @ baseclient.js:169
Hub2.bindClient @ hub.js:55
initAndBind @ sdk.js:27
init @ sdk.js:92
(anonymous) @ index.tsx:20
(anonymous) @ index.tsx:73
(anonymous) @ index.tsx:73
scheduler.production.min.js:12 Uncaught TypeError: Cannot read properties of undefined (reading 'lat')
    at LatLngBounds.contains (leaflet-src.js:1263:17)
    at FilteredPointLayer (FilteredPointLayer.ts:28:26)
    at MapPanel.tsx:309:37
    at fk (react-dom.production.min.js:262:359)
    at exports.unstable_runWithPriority (scheduler.production.min.js:18:343)
    at gg (react-dom.production.min.js:122:325)
    at Oj (react-dom.production.min.js:261:308)
    at react-dom.production.min.js:261:215
    at V (scheduler.production.min.js:16:224)
    at MessagePort.F.port1.onmessage (scheduler.production.min.js:12:346)
contains @ leaflet-src.js:1263
FilteredPointLayer @ FilteredPointLayer.ts:28
(anonymous) @ MapPanel.tsx:309
fk @ react-dom.production.min.js:262
exports.unstable_runWithPriority @ scheduler.production.min.js:18
gg @ react-dom.production.min.js:122
Oj @ react-dom.production.min.js:261
(anonymous) @ react-dom.production.min.js:261
V @ scheduler.production.min.js:16
F.port1.onmessage @ scheduler.production.min.js:12
trycatch.js:152          POST https://api.amplitude.com/ net::ERR_BLOCKED_BY_CLIENT
(anonymous) @ trycatch.js:152
(anonymous) @ instrument.js:248
Request2.send @ amplitude.esm.js:1717
sendEvents @ amplitude.esm.js:4070
_sendEventsIfReady @ amplitude.esm.js:2757
_logEvent @ amplitude.esm.js:3651
logEvent @ amplitude.esm.js:3784
logEvent @ amplitude.esm.js:3729
(anonymous) @ AmplitudeAnalytics.ts:61
logEvent @ AmplitudeAnalytics.ts:59
logEvent @ AnalyticsMetricsCollector.ts:17
initialized @ AnalyticsMetricsCollector.ts:23
(anonymous) @ index.ts:168
emit @ index.js:181
ws.onmessage @ FoxgloveClient.js:56
fetch.js:49          POST https://o530369.ingest.sentry.io/api/5649767/store/?sentry_key=a9a6c2015e1c4aec8aaa9d34014a1078&sentry_version=7 net::ERR_BLOCKED_BY_CLIENT`

foxymiles, 2022-04-26:
Thank you for the extra detail. Would it be possible for you to show me the entire GeoJSON message you're sending, including the definition of the coordinates?

tanpinsiang, 2022-04-26:
This is the [code](https://gist.github.com/tanpinsiang/99cdc525baca3c415f5d4835d0f8fbaa.js) I am using to send the websocket packet.
This is the GeoJSON. Thanks again.
```json
{
    "type":"FeatureCollection",
    "features":[
        {
            "type":"Feature",
            "properties":{},
            "geometry":{
                "type":"LineString",
                "coordinates":[
                    [
                        2.42477416992186,
                        1.801374964252865
                    ],
                    [
                        2.42082595825194,
                        1.7846897817763
                    ],
                    [
                        2.4422836303711,
                        1.78292608704408
                    ]
                ]
            }
        },
        {
            "type":"Feature",
            "properties":{},
            "geometry":{
                "type":"Polygon",
                "coordinates":[
                    [
                        [
                            2.44314193725585,
                            1.77342854582093
                        ],
                        [
                            2.43953704833984,
                            1.76596533600783
                        ],
                        [
                            2.4264907836914,
                            1.7694934927041
                        ],
                        [
                            2.42700576782227,
                            1.77817746896081
                        ],
                        [
                            2.44314193725585,
                            1.77342854582093
                        ]
                    ]
                ]
            }
        },
        {
            "type":"Feature",
            "properties":{
                "marker-color":"#7e7e7e",
                "marker-size":"medium",
                "marker-symbol":"1"
            },
            "geometry":{
                "type":"Point",
                "coordinates":[
                    2.43284225463867,
                    1.78943798147498
                ]
            }
        }
    ]
}
```

foxymiles, 2022-04-26:
@tanpinsiang Thanks for the code example. I think your issue is that the GeoJSON all needs to be under a `geojson` key as shown in the message definition here:

https://foxglove.dev/docs/studio/messages/geojson

Can you please try moving your data under that toplevel key?

foxymiles, 2022-04-26:
Ok good looks like we're making progress.

Currently the map panel also expects one non-GeoJSON point to use as the center. Could you please also try sending one point as a single point of type `foxglove.LocationFix` to set the map center?

tanpinsiang, 2022-04-26:
OK. This is my [code](https://gist.github.com/tanpinsiang/99cdc525baca3c415f5d4835d0f8fbaa) to send both `foxglove.LocationFix` and `foxglove.GeoJSON`. It this the correct way to send both channel together? The map panel is working but it is blinking between the two message.

foxymiles, 2022-04-26:
Yes that looks correct. The map panel will continue to show the most recently received message on the topics so you don't need to re-publish the message unless it's changed. That should eliminate the flickering.

If you have any followup questions could you please post them on our slack at https://foxglove.dev/slack?

tanpinsiang, 2022-04-26:
Awesome! Thanks!

PR #3254 (2022-04-25): Fix geojson marker images
**User-Facing Changes**
This fixes the missing marker images in GeoJSON layers on the map panel.

**Description**
Leaflet asset location and Webpack do not work well out of the box. We have to override the default leaflet icon to use the correct asset paths to our images.

See https://github.com/Leaflet/Leaflet/issues/4968#issuecomment-264311098

[image]

Fixes #3248