foxglove/studio #3260: Add the file path of the currently open data source to the PanelExtensionContext

f-o-w-l, 2022-04-25:
I would love access to the full file path of the currently open data source from a custom extension, so that I can run logic that also reads the source file from an extension.

f-o-w-l, 2022-05-06:
Instead of a file path actually, it would be even better as a `File` object, given that a `File` is required for [rosbag.js](https://github.com/cruise-automation/rosbag.js#opening-a-new-rosbag-reader) for parsing in the browser =)

defunctzombie, 2022-05-07:
Currently, the architecture of Studio is such that individual panels (and panel extensions) do not handle the data source. They are provided messages from the data source from the current playback (or even the entire range with the allMessages field). This is so that a panel, once authored, works across live and recorded data without having to handle such logic itself.

Could you elaborate more on what the logic is you want to run in the extension that needs access to the file? Maybe this is something that can be provided via the extension api rather than requiring the panel to re-parse the bag file.

f-o-w-l, 2022-05-07:
> Could you elaborate more on what the logic is you want to run in the extension that needs access to the file?

No problem, I need to search all of the messages on a given topic for some data, so it made much more sense to me to load the file through `rosbag.js` than it did to run `context.seekPlayback` millions of times for every timestamp.

It sounds like `allMessages` is exactly what I'm looking for, however GitHub says "We couldn’t find any code matching 'allMessages' in [foxglove/studio](https://github.com/foxglove/studio)." Where can I find that field? Thank you very much for your help!

f-o-w-l, 2022-05-11:
> I had mistakenly remembered it as allMessages, it's called "allFrames".

Thanks so much!

I do wonder if in the future I might run into a false negative trying to search `allFrames` for data on a topic in a very large and long .bag. It could be helpful in that case to have access to the open file size or total duration, to detect whether I should use `allFrames` or `rosbag.js` for parsing. For now though, `allFrames` has been working well and I'm happy with closing this issue.

f-o-w-l, 2022-05-27:
Hey, I'm re-opening this issue due to these two problems I've ran into trying to use `allFrames`.

1. The main one is simply that Foxglove will always crash before it will stop attempting to populate `allFrames` if you subscribe to a topic with many messages. It is supposed to be best-effort list to 1GB of memory, but it seems that it never stops loading more messages if there are more to load.
2. The other concern, that wouldn't be an issue if 1. wasn't also an issue, is that it doesn't seem like `allFrames` can be conditionally watched. To reproduce:
    - Add a console.log("Panel rendered") in the body of the entry panel component receiving the `PanelExtensionContext` (logging on every re-render)
    - Conditionally call `context.watch("allFrames")` in the `useLayoutEffect`, eg. `if (someBoolean) context.watch("allFrames")` and adding `someBoolean` to the `useLayoutEffect` dependency list
    - Change `someBoolean` to false from somewhere in the extension
    - Subscribe to a new topic/topics with over 1GB of messages
    - Observe that "Panel rendered" will be logged over and over until Foxglove crashes, even though `context.watch("allFrames")` was not called in the most recent `useLayoutEffect` call. Even though `allFrames` is not included in the renderState -- `renderState.allFrames` is correctly undefined after the new subscription(s) -- it seems to be getting populated in the background anyway.

When dealing with topics that don't have thousands of messages, `allFrames` has been working just fine. Thank you very much for your help!

f-o-w-l, 2022-05-27:
Opened #3463 instead.

closed 2022-05-27 by f-o-w-l