foxglove/studio #3272: Problem runnign Foxglove

BorisKontorovich, 2022-04-27:
**Description**
My ROS2 package crashes when running with FoxGlove

- Version: v1.9.0
- OS: Ubuntu 20.01

**Steps To Reproduce**

We are running Intel RealSense ROS2 wrapper publishing raw image with just one camera.

**Expected Behavior**
Foxglove can display published images

**Actual Behavior**
Intel RealSense Wrapper crashes a few seconds after FoxGlove is started.
It produces this error:

[cam-2] python3: /tmp/binarydeb/ros-galactic-cyclonedds-0.8.0/src/core/ddsi/src/ddsi_entity_index.c:411: entidx_lookup_proxy_reader_guid: Assertion `is_reader_entityid (guid->entityid)' failed.
[ERROR] [cam-2]: process has died [pid 3848, exit code -6, cmd '/home/sh/ws/flir-wrapper_ws/install/flir_wrapper/lib/flir_wrapper/cam --ros-args -r __node:=cam_727 -r __ns:=/cam_727 --params-file /tmp/launch_params_5fvheszt'].

defunctzombie, 2022-04-27:
@BorisKontorovich This looks like an error with your ROS2 node and not an error from Studio. I'd recommend looking into why the node is crashing as this would be specific to the node's logic. Maybe open an issue with the authors of the Intel RealSense ROS2 Wrapper with your above trace.

BorisKontorovich, 2022-04-27:
Can you give me some ideas how I can try foxglove with a camera?

BorisKontorovich, 2022-04-28:
Anything?

jtbandes, 2022-04-29:
Are you able to view raw data with `ros2 topic echo /my_camera_topic`?

@jhurliman Could you look into this as it appears to be DDS related? Is there a chance our RTPS implementation is generating some kind of id wrong, or doing something that triggers a bug in cyclonedds?

Looks like the assertion is here:

https://github.com/eclipse-cyclonedds/cyclonedds/blob/3c81d2856aa088e48d2e9f2365c53c2591ee6fdb/src/core/ddsi/src/ddsi_entity_index.c#L411

And the `is_reader_entityid` function:

https://github.com/eclipse-cyclonedds/cyclonedds/blob/3c81d2856aa088e48d2e9f2365c53c2591ee6fdb/src/core/ddsi/src/q_entity.c#L211-L221

jhurliman, 2022-04-29:
I can't reproduce this issue yet, but I've found at least one bug related to `DATA_FRAG` handling (large payloads such as image topics). I'm looking into it now.

jhurliman, 2022-05-03:
Although I haven't been able to reproduce this specific crash, I did find some issues with our protocol handling related to large messages while testing the ros2_realsense_camera node. They have been fixed in https://github.com/foxglove/rtps/pull/18 which will bubble up to a Studio fix soon.

BorisKontorovich, 2022-05-04:
I tried the settings in foxglove/rtps#18<https://github.com/foxglove/rtps/pull/18> but it still runs into the same problem.
[https://opengraph.githubassets.com/552c744bda0a11ded849ced698ac6ae02f8b70aa52b4c12dc0e02c5787c5ae03/foxglove/rtps/pull/18]<https://github.com/foxglove/rtps/pull/18>
Fixes for receiving large messages by jhurliman · Pull Request #18 · foxglove/rtps<https://github.com/foxglove/rtps/pull/18>
NOTE: Receiving large messages such as 1MB raw image frames may not be possible with the default Linux networking receive buffer size of 256KB, depending on CPU speed / contention / network speed /...
github.com

Boris K.

jhurliman, 2022-05-05:
@BorisKontorovich the fixes made in the linked PR are needed as well. They will be included in the next release (1.10.0) which should go out on Monday May 9th.

HemaZ, 2022-05-24:
@jhurliman 
I'm still having this issue on 
Foxglove Studio version 1.12.0

Playing a ROS2 bag and trying to use foxglove to visualize the camera images. 
ROS2 bag crashes with the following error 
```
python3: /tmp/binarydeb/ros-galactic-cyclonedds-0.8.0/src/core/ddsi/src/ddsi_entity_index.c:411: entidx_lookup_proxy_reader_guid: Assertion `is_reader_entityid (guid->entityid)' failed.
```

using RViz2 for visualizing works with no issue. So i think the issue might be related to Foxglove rtps.

defunctzombie, 2022-06-14:
@jhurliman What's the status of this issue? Is it one that impacts anyone trying to view camera images on ROS2 systems? I'm not seeing other reports of problems from others using ROS2 with studio.

jhurliman, 2022-06-14:
@defunctzombie I've been unsuccessful in root causing this issue so far, it seems to affect CycloneDDS under some circumstances that I can't reproduce.

bertaveira, 2022-06-28:
I get the same problem on Galactic (Ubuntu 20.04) using the Ouster driver: https://github.com/ros-drivers/ros2_ouster_drivers

Steps:
1. Start driver and wait for it to connect to lidar
2. open foxglove if not already opened and go on the 3D panel. Click on the points topic (pointcloud2) to display it on foxglove
3. it should crash the Ouster driver

bertaveira, 2022-06-29:
I managed to create a replicable scenario without the need for specific hardware such as a lidar. It uses the same Ouster lidar driver but it reads from a PCAP file instead of the actual sensor. The Ouster driver has a separate mode to run with recorded PCAP files but I have verified it still crashes and the same behaviour is present. 

You will require a metadata file and a PCAP file both of which I got blocked to upload here so here is this link instead: https://drive.google.com/drive/folders/1WXLTntiAHcr_mCFAgR7GjQl3VH5BR5dO?usp=sharing

**Note:** The recording file is only a bit over 1min long

Here is the steps to replicate:
1. Clone the ouster driver repo in a computer with ROS2 (galactic recommended since the problem seems related to cyclonedds): https://github.com/ros-drivers/ros2_ouster_drivers
2. Follow instructions at the end of the readme to setup the TinsDriver mode so it will read from the recording and not crash: https://github.com/ros-drivers/ros2_ouster_drivers#usage-with-tins-based-driver
- I have generated a metadata file for the sensor recorded that should be used instead of the default ones in the repo
- You might need to go to `sudo -s` and then source ros2 and only then launch the node since the libtins requires sudo permissions or it will throw an error
3. After changing the ethernet device, building and sourcing run it 
```
ros2 launch ros2_ouster tins_driver_launch.py metadata_filepath:=/path/to/file/OS1_64_BH_metadata.json
```
4. Start pcap replay. If not installed then `sudo apt-get install tcpreplay`
```
sudo tcpreplay --intf1=[ethernet device] ouster_lidar_rec.pcap
```
5. Verify that it is running and publishing the pointcloud correctly (with rviz2 or rqt or anything else)
6. Open foxglove and open a 3D panel. Select the topic /points
7. It should crash the ros2_ouster node

defunctzombie, 2022-07-29:
Sorry we've been dormant on this issue. For now we recommend using the rosbridge connection option when trying to connect to ROS2 robots. The rosbridge option tends to work more reliably across the different ros2 variants and also works with custom messages whereas the native ros2 connection in studio does not work with custom messages.

jhurliman, 2022-09-13:
I'm trying to reproduce this issue given the latest instructions, and I don't see the crash but I do see error messages on the `ros2 bag play` side and Studio fails to subscribe to `/ouster/points` while other topics appear to work fine. I built ROS2 Galactic from source in a container using <https://gist.github.com/jhurliman/d1ad9e2c78bb81adfd26606960551e7f>, started playback of the provided MCAP file, connected the latest desktop release version of Studio, and when attempting to subscribe to `/ouster/points` the `ros2 bag play` terminal prints a series of errors that look like this:

```
1663089166.116827 [0]     recvUC: malformed packet received from vendor 1.16 state parse:nackfrag <52545053 02010110 63e380f2 69edf702 c8770214 0e010c00 0110f9e5 d365ca5e @0x24 12013c00 00000000 031e0000 00000000 1c150000 00000000 00010000 ffffff7f> (note: maybe partially bswap'd) {{12,1,60},0,1e03,5404,0,256}
1663089166.116837 [0]     recvUC: malformed packet received from vendor 1.16 state parse:nackfrag <52545053 02010110 63e380f2 69edf702 c8770214 0e010c00 0110f9e5 d365ca5e @0x24 12013c00 00000000 031e0000 00000000 1c150000 00000000 00010000 ffffff7f> (note: maybe partially bswap'd) {{12,1,60},0,1e03,5404,0,256}
1663089166.116854 [0]     recvUC: malformed packet received from vendor 1.16 state parse:nackfrag <52545053 02010110 63e380f2 69edf702 c8770214 0e010c00 0110f9e5 d365ca5e @0x24 12013c00 00000000 031e0000 00000000 1c150000 00000000 00010000 ffffff7f> (note: maybe partially bswap'd) {{12,1,60},0,1e03,5404,0,256}
1663089166.116863 [0]     recvUC: malformed packet received from vendor 1.16 state parse:nackfrag <52545053 02010110 63e380f2 69edf702 c8770214 0e010c00 0110f9e5 d365ca5e @0x24 12013c00 00000000 031e0000 00000000 1c150000 00000000 00010000 ffffff7f> (note: maybe partially bswap'd) {{12,1,60},0,1e03,5404,0,256}
```

While it's not a crash, this should be enough to go off for now.

jhurliman, 2022-09-13:
Any Linux users running into this issue or other issues with the ROS2 native connection, please try following the `sysctl` tuning steps at https://github.com/foxglove/rtps#notes and report back if this improves the behavior.

HemaZ, 2022-09-14:
> Any Linux users running into this issue or other issues with the ROS2 native connection, please try following the `sysctl` tuning steps at https://github.com/foxglove/rtps#notes and report back if this improves the behavior.

Tried the steps in the link , still getting crash on ROS2 Galactic ubuntu 20.04

```
[INFO] [1663148281.258157400] [rosbag2_storage]: Opened database 'kitti_2011_10_03_drive_0027_synced2_readme/kitti_2011_10_03_drive_0027_synced2_readme.db3' for READ_ONLY.
[INFO] [1663148281.259853570] [rosbag2_storage]: Opened database 'kitti_2011_10_03_drive_0027_synced2_readme/kitti_2011_10_03_drive_0027_synced2_readme.db3' for READ_ONLY.
python3: /tmp/binarydeb/ros-galactic-cyclonedds-0.8.0/src/core/ddsi/src/ddsi_entity_index.c:411: entidx_lookup_proxy_reader_guid: Assertion `is_reader_entityid (guid->entityid)' failed.
Aborted (core dumped)
```

jhurliman, 2022-09-15:
The latest studio `main` branch includes a new version of our RTPS library (PR at https://github.com/foxglove/studio/pull/4438). Unfortunately, I haven't been able to consistently repro the crash in my environment so I can't confirm if it fixes the reported issue. Still, it does improve protocol compliance in a few places, so fingers crossed. If someone seeing this crash can test out a desktop build of `main` and report back it would be much appreciated.

HemaZ, 2022-09-16:
> The latest studio `main` branch includes a new version of our RTPS library (PR at #4438). Unfortunately, I haven't been able to consistently repro the crash in my environment so I can't confirm if [...]

Can you point me to the latest main build? or how can I build it?

jhurliman, 2022-09-16:
@HemaZ we don't publish continuous/nightly builds of the desktop app, only the web version. For desktop, you can follow the instructions at https://github.com/foxglove/studio#contributing to clone the repository and build from source.

defunctzombie, 2022-09-16:
@HemaZ Here's a dev build of latest main: https://drive.google.com/file/d/1mLkw_T3OfeZiX7E2XhRPvcKWmDHp7Sys/view?usp=sharing You can give this a try or wait until we make a new release next week and try that.

HemaZ, 2022-09-17:
> @HemaZ Here's a dev build of latest main: https://drive.google.com/file/d/1mLkw_T3OfeZiX7E2XhRPvcKWmDHp7Sys/view?usp=sharing You can give this a try or wait until we make a new release next week and try that.

Tried this release. still crashing.

defunctzombie, 2022-09-26:
Using the latest version of Studio 1.26.0.

I ran the above steps for `ros2 bag play` with the referenced `cfs22_mar_track4_fast.mcap` file on a Focal VM with the latest ros2 galactic `ros2-testing` packages (http://packages.ros.org/ros2-testing/ubuntu focal main). I was able to subscribe to `/ouster/points` and receive data. The `ros2 bag play` process did not crash.

defunctzombie, 2022-09-26:
@bertaveira Are you able to try again with the latest packages from ros2-testing? So far I've not been able to reproduce any crash using my VM.

defunctzombie, 2022-10-09:
@BorisKontorovich Are you still able to reproduce this issue with the latest release of Studio? Are you able to reproduce if you update your ROS galactic packages to the latest release from ros2-testing?

defunctzombie, 2022-10-13:
I was able to reproduce on a ubuntu linux machine.

jhurliman, 2022-10-25:
I've moved this ticket to the @foxglove/rtps library where the bug exists, see <https://github.com/foxglove/rtps/issues/22>. We'll keep this Studio ticket open as well until the issue is resolved or the ROS 2 native connector is deprecated.

defunctzombie, 2022-10-31:
@jhurliman now that you can reproduce this are you able to create a fix?

jhurliman, 2022-10-31:
@defunctzombie no, I have not had any luck in tracking down the root cause even though I can reproducibly crash a ROS 2 stack running the CycloneDDS RMW. If myself or anyone picks up @foxglove/rtps development, I think compiling CycloneDDS + rmw_cyclonedds from source and breakpoint-debugging or print-debugging would be a helpful next step.

felixf4xu, 2022-11-13:
I'd like to share what I have found so far:

during the ros2 node crash, the call stack is:
```
libddsc.so.0!entidx_lookup_proxy_reader_guid(const struct entity_index * ei, const struct ddsi_guid * guid) (\home\someone\github\cyclonedds\src\core\ddsi\src\ddsi_entity_index.c:412)
libddsc.so.0!handle_NackFrag(struct receiver_state * rst, ddsrt_etime_t tnow, const NackFrag_t * msg, SubmessageKind_t prev_smid, struct defer_hb_state * defer_hb_state) (\home\someone\github\cyclonedds\src\core\ddsi\src\q_receive.c:1556)
libddsc.so.0!handle_submsg_sequence(struct thread_state1 * const ts1, struct ddsi_domaingv * gv, ddsi_tran_conn_t conn, const ddsi_locator_t * srcloc, ddsrt_wctime_t tnowWC, ddsrt_etime_t tnowE, const ddsi_guid_prefix_t * const src_prefix, const ddsi_guid_prefix_t * const dst_prefix, unsigned char * const msg, const size_t len, unsigned char * submsg, struct nn_rmsg * const rmsg, _Bool rtps_encoded) (\home\someone\github\cyclonedds\src\core\ddsi\src\q_receive.c:3013)
libddsc.so.0!do_packet(struct thread_state1 * const ts1, struct ddsi_domaingv * gv, ddsi_tran_conn_t conn, const ddsi_guid_prefix_t * guidprefix, struct nn_rbufpool * rbpool) (\home\someone\github\cyclonedds\src\core\ddsi\src\q_receive.c:3295)
libddsc.so.0!recv_thread(void * vrecv_thread_arg) (\home\someone\github\cyclonedds\src\core\ddsi\src\q_receive.c:3510)
libddsc.so.0!create_thread_wrapper(void * ptr) (\home\someone\github\cyclonedds\src\core\ddsi\src\q_thread.c:241)
libddsc.so.0!os_startRoutineWrapper(void * threadContext) (\home\someone\github\cyclonedds\src\ddsrt\src\threads\posix\threads.c:165)
libpthread.so.0!start_thread(void * arg) (\build\glibc-SzIz7B\glibc-2.31\nptl\pthread_create.c:477)
libc.so.6!clone() (\build\glibc-SzIz7B\glibc-2.31\sysdeps\unix\sysv\linux\x86_64\clone.S:95)
```
in function of `handle_submsg_sequence()`, the `sm->smhdr.submessageId` is `(enum SubmessageKind)SMID_NACK_FRAG = 18`.

in function of `int is_reader_entityid (ddsi_entityid_t id)`, id.u is: 0x0

normal case:
```
sm->smhdr.submessageId is: 14
sm->smhdr.submessageId is: 6
is_reader_entityid returns 4, id.u is: 0x1404
is_reader_entityid returns 4, id.u is: 0x304
is_reader_entityid returns 4, id.u is: 0x404
sm->smhdr.submessageId is: 14
```
crash case:
```
sm->smhdr.submessageId is: 18
is_reader_entityid returns 0, id.u is: 0x0
```

from code `src.entityid = msg->readerId;`, I can find that `msg->readerId.u` is 0 then the node crashes, only at case of SMID_NACK_FRAG.

I may need help from here.

jtbandes, 2022-11-30:
Going forward, we plan to deprecate and remove the ROS 2 native connection (https://github.com/foxglove/studio/issues/4900). We are working on a new [foxglove_bridge](https://github.com/foxglove/ros-foxglove-bridge), which will provide improved performance and reliability over the Rosbridge and native connections. You can already use the new bridge today (see the README). Stay tuned for upcoming announcements about the new bridge!

closed 2022-11-30 by jtbandes