foxglove/studio #3274: Loading layouts sometimes takes a long time

jessicaaustin, 2022-04-27:
**Description**

When I open the Layouts side panel, sometimes it takes a few seconds for the layouts to load. Today, things were especially bad, with loading taking 30s or more at times.

https://user-images.githubusercontent.com/1159930/165590102-fe5833f9-9f52-4180-b7ee-24e55a058df3.mp4

This also happens after the Layout panel loads for the first time -- it looks like it gets refreshed periodically, and while it's refreshing, I can't switch to another layout.

- Version: 1.7.0
- OS: Ubuntu

**Steps To Reproduce**

I don't have nice clean steps to replicate here with a dummy setup, so I'll try to describe my real setup and what I'm seeing as best as possible.

* I am logged in with an account, and I have about ~50 personal layouts and about ~10 team layouts. Some of these layouts are fairly complicated.
* It looks like there's a request for `/layouts` that happens periodically, all of those are coming through quickly so it doesn't seem like a delay due to a remote server
   * [image]
* I'm not seeing any obvious errors in the Console when this happens, though I do see some warnings, dunno if they're related
   * [image]

The one thing I think is influential is the complexity of the currently loaded layout, and (possibly) the overall load on the computer too. For example, if I load a complex layout with lots of plots, then in general Studio becomes sluggish, including the processing of Layouts. Here's one example where i took the demo layout and duplicated a bunch of panels and added a plot panel. I'm able to get Studio processes to use a lot of CPU this way, and then when I open the Layouts side panel, it takes ~5s to load.

[image]

I understand that high load on the running machine is out of your hands, but this can lead to frustration because if I made a layout "too complicated" then I can't switch out of it without having to reload everything. Perhaps one "fix" might be to have a setting about how often to refresh layouts, so I could lower that refresh rate in this situation? Or to not freeze the layouts panel and allow you to switch while it's being loaded? For example I tried logging out of my account so it would stop loading, but then all my layouts were gone. :shrug: 

I have a few videos of this behavior on the real setup if it's helpful, DM me in slack if you'd like to see them.

foxymiles, 2022-04-27:
Thanks again for taking the time to write such a careful & detailed bug report!

jessicaaustin, 2022-04-28:
Yeah I should be clear I don't think this is a bug necessarily, more of a UX issue. If I'm pushing the limits of Studio and the laptop, I'd like to be able to turn down the frequency of Layout rendering, similar to how I can turn down the Message Rate in the settings, so that I can still navigate layouts when resources are low. (Or some other approach to deal with it, this is just an idea) Thanks for considering!

defunctzombie, 2022-04-29:
While not all aspects of this might be a bug - I think its a bug that the layout list isn't showing even tho its already been cached locally (from what I've read in your report) - or that you can't select a layout while the list is loading in the background. It didn't sound to me like Studio itself was under load from playback at the same time.

jessicaaustin, 2022-04-29:
> you can't select a layout while the list is loading in the background

Yes, this appears to be what is happening. I see the network call to `/layouts`, then I see the spinner at the top of Layouts start spinning, and while it's spinning if I click another layout it doesn't load until the spinner stops. I assume the spinner indicates layouts are being refreshed.

foxymiles, 2022-05-16:
@jessicaaustin What kinds of plots are you doing? Are they plots of live data with a steadily accumulating buffer?

foxymiles, 2022-05-16:
I can repro this with a simple sinusoid source over rosbridge and about 8 simultaneous plots. The whole UI gets sluggish and it's particularly noticeable with layouts since they're refreshing at a fixed frequency and the app spends an increasingly large portion of the time between each refresh processing the layout data from the refresh so that the layout panel is constantly busy.

foxymiles, 2022-05-16:
Also do you tend to set an x range for your plots or just graph the entire timerange?

[image]

jessicaaustin, 2022-05-17:
@foxymiles in general I'm setting an X range, because I have noticed that that helps performance. To give a concrete example, one layout that usually has issues has 6 plots, 5 of which have 30s window and one which I just realized is still set to `auto`. It also has 9 Raw Message panels and one panel that's a custom extension.

If it's helpful I can DM you some screenshots in slack but it sounds like you're able to reproduce the issue?

foxymiles, 2022-05-17:
Yeah I'm able to reproduce the issue thanks. I think like you said it has a lot to do with studio struggling to stay responsive in those scenarios. We've discussed a few different strategies for making the plot panel more efficient which it why it's helpful to have a clearer idea of how people are using it. It's interesting that even with an X range configured you're still seeing this.

jessicaaustin, 2022-06-03:
Any update on this issue? It's still a pain point for us. Thanks!

foxymiles, 2022-06-03:
We've discussed various approaches to improving the performance of the plot panel in particular and I think that will be part of this solution. Let me see if I can improve our layout syncing behavior in the meantime so it's more usable when studio is loaded.

foxymiles, 2022-06-03:
@jessicaaustin Can you try this branch and see if the behavior is any better in your high load scenarios? https://github.com/foxglove/studio/pull/3489

foxymiles, 2022-06-06:
@jessicaaustin I'm hoping this PR will help in your situation but I'm waiting until you confirm before merging. https://github.com/foxglove/studio/pull/3489

jessicaaustin, 2022-06-06:
@foxymiles ty, I will pull and test now

defunctzombie, 2022-06-08:
@jessicaaustin Related question to this - is using the sidebar slow for anything else (other than layouts) when you have your plot layout open? For example - if you have the plot layout open and go to add a new panel via the `Add panel` sidebar is that sidebar slow to open? Another example - if you open the panel settings sidebar - is that sidebar slow to render/open and use?

We're trying to isolate if this is a perfomance slowdown overall or if it is specific to using and changing layouts.

foxymiles, 2022-06-08:
Looking more closely at this I think this is unrelated to syncing. The issue is that we reload the full list of layouts every time the layout browser is opened. On a heavily loaded machine with a lot of layouts this can be slow.

I'm guessing we need to do some smart caching here instead of reloading the layouts every time.

foxymiles, 2022-06-09:
For another repro try using this as a data source over rosbridge:

https://github.com/foxglove/ros1-bench-workspace

And this attached layout:
[many plots.json.zip](https://github.com/foxglove/studio/files/8874077/many.plots.json.zip)

jtbandes, 2022-06-10:
@jessicaaustin Could you try this alternative branch out? #3571

jessicaaustin, 2022-06-11:
@jtbandes MUCH better.

I tested by checking out that branch, yarn serve + yarn start. I don't know how to login with my account when it's the dev environment, so I couldn't test remote layouts or syncing with that. But I did create a big layout with like 2 dozen plots and a bunch of tabs, duplicated it so I had ~30 layouts like that (for reference in my real account I have more like 80+ layouts), and played back data so they were all running. When I switch to the Layouts tab, there's no longer a spinner or any delay at all. I did some limited functional testing, like updating layouts, closing and re-opening and verifying it saved, didn't see any issues.

If you have any tips on how I can create an account at https://console-dev.foxglove.dev/signin to sync remote layouts I can try that too.

jessicaaustin, 2022-06-11:
> Related question to this - is using the sidebar slow for anything else (other than layouts) when you have your plot layout open?  For example - if you have the plot layout open and go to add a new [...]

@defunctzombie this question might be outdated now, but no I don't see anything else slowing down on the sidebar. When Studio is under heavy load I can see delays interacting with specific panels though -- for example in a plot panel editing the topic that is being plotted. Not sure there is any way around that though.

PR #3571 (2022-06-13): Add cache for layout storage
**User-Facing Changes**
Improved performance of the Layouts sidebar with more caching.

**Description**
Alternative to #3557
Fixes #3274
Revival of/successor to code added in #1206 & removed in #1679

Marking as draft because we need to discuss the implications on using multiple Studio windows/tabs/instances. Currently the cache will not be reloaded during the whole session, which means changes in the underlying storage from another instance of the app will not be picked up. Is this acceptable?
commit 5694aea0 (2022-06-13): Add cache for layout storage

**User-Facing Changes**
Improved performance of the Layouts sidebar with more caching.

**Description**
Adds a new cache layer.

Alternative to #3557
Fixes #3274
Revival of/successor to code added in #1206 & removed in #1679