foxglove/studio #3277: Settings sidebar issues

jhurliman, 2022-04-28:
I've been implementing the settings sidebar in the 3d panel and overall it looks great! Here are the issues I've encountered:

- [ ] The default sidebar width is not wide enough for almost any of the topic settings labels. they are cut off with ellipses
- [ ] `"gradient"` isn't usable yet, value should have type `[string, string]` and the control be a two-step gradient picker, not a color map dropdown
- [ ] `"number"` would be much easier to implement with `min`, `max`
- [ ] `"number"` could have better UX if the arrow buttons and arrow keys worked when there is only a placeholder value (parse as a number or fall back to 0)
- [ ] `"select"` setting value: `""` doesn't select the option entry with value set to `""`
- [ ] Need a visibility toggle control and some field or event for visibility changes
- [ ] The single `"color"` is confusing, i need separate `"rgb"` and `"rgba"` input types
- [ ] `"color"` could benefit from a placeholder color option
- [ ] We might need a validation step and error highlighting (color input red?). example: type only `e` in a number input

foxymiles, 2022-05-02:
@jhurliman The number of digits shown seems like something we should leave up to panel implementers to implement when they build the settings tree.

foxymiles, 2022-05-02:
For rgba colors do you want them as css `rgba()` style strings or as 8-digit hex numbers?

foxymiles, 2022-05-02:
Actually it looks like our current color picker widget doesn't handle 8 digit hex numbers well so for now it's just going to be `rgba()`.

jhurliman, 2022-05-17:
@foxymiles where do I put that function call? If there was a format callback attribute on number fields I could use it, but I don't see how to incorporate that currently.

foxymiles, 2022-05-18:
Just put in in the code where you build the settings tree in the `value` attribute for the field.

https://github.com/foxglove/studio/blob/0b14b25e9e9c7d170438e7c9b32556c72af21773/packages/studio-base/src/panels/ThreeDeeRender/settings.ts#L195-L202

jhurliman, 2022-05-18:
@foxymiles this is too late, it means I would be storing a value with higher precision than what is shown in the sidebar. I was hoping for an attribute that would prevent me from ever receiving a `SettingsTreeAction` with a high precision value.

foxymiles, 2022-05-18:
You could also catch this in your action handler and prevent any high precision value from ever getting set too.

I can easily add this on the editor side but what should it do exactly? Round up, down, truncate? How about negative numbers?

jhurliman, 2022-05-18:
This is what copilot suggested:

```
function limitPrecision(x: number, digits = 3): number {
  const factor = Math.pow(10, digits);
  return Math.round(x * factor) / factor;
}
```

closed 2022-05-18 by jhurliman