foxglove/studio #3283: Adding decay time makes point cloud disappear

amacneil, 2022-04-28:
**Description**
I have a file with sensible receive times in a PointCloud2 message, and incorrect "header stamp" values.

Playing back this file works fine. However, if I add "decay" to the topic in 3d panel, the 3d panel goes blank.

[image]

Apparently this is due to the invalid header stamp times. If we are playing back by receive time, then we should decay by receive time, and header stamp time shouldn't factor in to it.

- Version: main
- OS: web

foxymiles, 2022-05-04:
Looks like this is a consequence of this change: https://github.com/foxglove/studio/pull/2986

defunctzombie, 2022-05-12:
@foxymiles Whats the next step here? I noticed an attempt at this in #3314 but that didn't work because it was applying too broadly. Is there a fix here that should be specific to point clouds?

@jhurliman it looks like this broke in #2986 - what do you think should be done to fix this?

foxymiles, 2022-05-13:
@defunctzombie Seems like there are people that understand this code much better than I do and have ideas on how it should be fixed so I'm probably not the best person to take this one up.

jhurliman, 2022-05-13:
I can take this one. Timestamp handling in Old3D is very brittle right now, hopefully I can come up with a surgical fix for this specific issue without additional side effects.

defunctzombie, 2022-06-13:
We've decided it is not worth investing in fixing this in old3d and focus on implementing this correctly in new3d.

PR #3755 (2022-06-30): [New3D] PointCloud and LaserScan decay
**User-Facing Changes**

- Implements the decay setting for PointCloud and LaserScan topics in `3D (Beta)` panel

**Description**

Fixes #3283. Implements decay for point clouds and laser scans. The original messages are not kept around (a consideration for per-point picking when that is implemented), only the GPU buffers.

Other fixes:

- Option to override setUsage() for DynamicBufferGeometry
- Don't expire transforms after 60s
- Dispose renderables in removeAllRenderables()
- Fix transform lookup when all TFs are after the search time but within maxDel [... cut]
closed 2022-06-30 by jhurliman