foxglove/studio #3284: Variables don't work in quoted topic names

jameskuszmaul-brt, 2022-04-28:
**Description**

I've got topic names that require escaping (which was solved by #2891), but I don't seem to be able to use variables in my topic names (I don't know if variables are supposed to be usable in normal topic names either, admittedly). I'd like to be able to do something like:

`"/\$machine_name/topic name".xyz`

**Steps To Reproduce**

Can provide a bag file as needed.

**Expected Behavior**

**Actual Behavior**

jtbandes, 2022-04-29:
Converting to a feature request – variables don't work in topic names currently, only filters.

jtbandes, 2022-09-27:
Related: https://github.com/foxglove/studio/issues/2907

jameskuszmaul-brt, 2022-09-27:
The glob pattern idea referenced in #2907 would also resolve the core use-case referenced here, although I could still generate some corner-cases where variable names in topics would be helpful.