foxglove/studio #3285: Studio freezes when trying to step forward towards the end of a rosbag

rgov, 2022-04-29:
**Description**

- Version: 1.9.0
- OS: macOS 12.3.1

**Steps To Reproduce**

Open a particular bag file, seek to the very end, keep stepping forward to try to find the last message.

**Actual Behavior**

UI becomes unresponsive. If Inspector is already open, then I see a very rapid log spew of identical messages like this:

[image]

defunctzombie, 2022-04-29:
Confirming I am able to reproduce. Looking into this.

defunctzombie, 2022-04-29:
@rgov I've identified the issue and created a fix in #3291. We'll get this in the next release on Monday.

PR #3291 (2022-05-03): Fix the tick handling for IterablePlayer

**User-Facing Changes**
When seeking near the end of the data, the app no longer freezes if there are no messages between the start of the seek and the end of the dataset time.

**Description**
When performing a _tick_ the iterable player would fail to track the correct _next_ start time for an iterator when the _tick_ needs to span two iterators (i.e. the current iterator ends but we still need to read more messages to finish the tick). This change fixes this logic to always set the next start time to the end of the iterator and also clear the lastMes [... cut]
commit 79ecaddb (2022-05-03): Fix the tick handling for IterablePlayer (#3291)

**User-Facing Changes**
When seeking near the end of the data, the app no longer freezes if there are no messages between the start of the seek and the end of the dataset time.

**Description**
When performing a _tick_ the iterable player would fail to track the correct _next_ start time for an iterator when the _tick_ needs to span two iterators (i.e. the current iterator ends but we still need to read more messages to finish the tick). This change fixes this logic to always set the next start time to the end of the iterator and also clear the [... cut]
closed 2022-05-03 by defunctzombie