foxglove/studio #3287: Foxglove Teleop properly publishes to ROS topic but robot doesn't respond

liamreneroy, 2022-04-29:
**Description**

I am trying to control my Jackal robot using the Teleop feature within Foxglove. To control the robot, you must publish to the ROS topic /cmd_vel

I am able to publish to the /cmd_vel topic using the Teleop feature within Foxglove. I've confirmed that Foxglove is in fact publishing to this topic by running "rostopic echo /cmd_vel". Even though Foxglove seems to publish to the topic correctly, the robot doesn't repond. 

NOTE that when publishing to the same topic /cmd_vel through the terminal using  "rosrun teleop_twist_keyboard teleop_twist_keyboard.py" the robot responds no problem. 

I can watch the terminal displaying "rostopic echo /cmd_vel" and go back and forth between publishing using teleop_twist_keyboard and Foxglove.. the robot only moves when I publish from teleop_twist_keyboard, even though the topic shows input from both teleop_twist_keyboard publishing & Foxglove publishing. 

- Version: 1.9.0
- OS: Ubuntu 20.04 w/ ROS Noetic

**Steps To Reproduce**
1. Setup Ubuntu 20.04
2. Install ROS Noetic & Gazebo 11 ~ http://wiki.ros.org/noetic/Installation/Ubuntu
3. Install Foxglove ~ https://foxglove.dev/download
4. Install Jackal gazebo ~ https://github.com/jackal/jackal_simulator 
5. In a new terminal: roslaunch jackal_gazebo empty_world.launch 
6. In a new terminal: rosrun teleop_twist_keyboard teleop_twist_keyboard.py
7. In a new terminal: rostopic echo /cmd_vel
8. Open Foxglove studio and setup Teleop controls to publish to /cmd_vel
9. Be confused why this isn't working.. :( 

**Expected Behavior**
Jackal robot should move whenever you publish to the topic /cmd_vel, regardless of whether it's through teleop_twist_keyboard or Foxglove.

**Actual Behavior**
Jackal robot only responds when publishing from teleop_twist_keyboard 

[image]

defunctzombie, 2022-04-29:
We've published v1.9.1. Please give that a try and see if it resolves the issue.

liamreneroy, 2022-05-02:
Does not seem to have resolved the issue after trying Foxglove v1.9.1

After further investigation using rqt_graph, it appears that Foxglove node is not properly publishing to the active /cmd_vel topic. 

I dont fully understand how that could happen, but when I run "rosrun rqt_graph rqt_graph" WITHOUT having "rostopic echo /cmd_vel" running, it appears that the Foxglove node is not linked with anything. (see photos 1 & 2)

After running "rostopic echo /cmd_vel", it appears that the Foxglove node only publishes to a duplicate /cmd_vel topic associated with the echo, and the messages are not picked up by the original /cmd_vel topic which is linked to the /twist_mux node that drives the Jackal robot (photos 3 & 4)

Any ideas of how to fix this or why it might be happening? 

[image]
[image]
[image]
[image]

liamreneroy, 2022-05-02:
As a further check, I also ran "rostopic hz /cmd_vel" 

No matter what publish rate I entered in Foxglove, "rostopic hz /cmd_vel" didn't pick up any published messages.

defunctzombie, 2022-05-03:
This issue might be related/similar to https://github.com/foxglove/studio/issues/3182

/cc @jtbandes we can join forces to debug this depending on where you are with the other issue.

defunctzombie, 2022-05-04:
Fixed by https://github.com/foxglove/studio/pull/3309

I've tested this fix and was able to get the robot to move in gazebo. This fix will be in the next release.

closed 2022-05-04 by defunctzombie

liamreneroy, 2022-05-14:
Hey guys! Thanks for attempting to resolve this issue. With the newest version of FoxGlove (1.10.1) I can get my robot moving in my Gazebo sim. Good stuff!

One issue still persists.. the Teleop buttons only seem to work half the time when I re-open a saved FoxGlove layout containing a TeleOp panel publishing to /cmd_vel.

A fix is if I open the TeleOp panel settings, and toggle the selected topic back and forth from /cmd_vel to another topic, then back to /cmd_vel. After that it starts working again. Any idea why this might be and if there's a way to make it work from start-up?

Cheers ~ Liam

[image]