foxglove/studio #3293: When a connection to a topic is reestablished, it should clear the errors

MathieuMarchand, 2022-04-30:
**Description**
When a node that publishes on a topic is killed or dies and a custom extension panel is subscribed to this topic, an error message is displayed on foxglove. But when the node is re-run and the topic is being used again, the error is not cleared even though the whole system is valid.

- Version: 20.04.4
- OS: Ubuntu

**Steps To Reproduce**
1. Subscribe to a topic on a custom extension published by a ros node
2. Kill the node that publishes on the topic
3. re-run the node 

[image]

**Expected Behavior**
When the connection to a topic is reestablished, the errors should be cleared

**Actual Behavior**
When the connection to a topic is reestablished, the errors are still there even though the topic is being used

jtbandes, 2022-05-18:
Are you using Rosbridge or a native (TCP) ROS connection?

MathieuMarchand, 2022-05-19:
The native ROS connection to ROS 1

PR #3451 (2022-05-25): Clear topic subscription errors when new messages are received
**User-Facing Changes**
None

**Description**
Currently once we encounter a subscription error on a topic on a ROS connection we display a player error but we don't clear this error once the subscription to that topic is working again. The solution here is to clear the error once we successfully receive a message on that topic again.

Fixes #3293
closed 2022-05-25 by foxymiles