foxglove/studio #3294: Plot does not correspond to latest message

rgov, 2022-05-01:
**Description**

- Version: 1.9.0
- OS: macOS 12.3.1

**Steps To Reproduce**

I opened a bag file (second in a series, `*_1.bag`) and seeked to close to the end. I opened a Plot panel and a Raw Message panel showing the same data.

**Expected Behavior**

All panels are in sync with the current playback position.

**Actual Behavior**

The plot of a particular value seems shifted. For instance the Raw Message panel shows 0.52 m depth while the plot shows 0.88 m depth at the same position.

To me it looks like the Plot panel is about 22 seconds early.

[image]

defunctzombie, 2022-05-02:
@rgov Is the plot series for `/ctd/depth.depth` using `header.stamp` time for plotting or `receive time`?

rgov, 2022-05-02:
`header.stamp`

defunctzombie, 2022-05-02:
Ok that makes sense. Here's the explanation.

When you plot data by `receive time` or `header.stamp` the `x-axis` represents the _time_. The difference is that when you plot using `receive time` the x-values from the `receive time` and when using `header.stamp` the x-values come from the message header stamp.

The confusion here is that the gray line x-value is always the receive time. So when it is shown on the plot, its x-value will be the receive time (remember the x-axis is _time_, not any specific time). And your points have x-values of header time. In your data these two are ~22 seconds apart. The message you see in the raw message panel is indeed the latest message for the playback time (you can see that in the receive time in the raw message panel), and the gray line and hover line of the plot occur at the x-axis value of the receive time. The tooltip also shows the values for the x-axis value when you hover, it just happens that those values come from being plotted with x-values that come from `header.stamp` rather than receive time.

The takeaway is that the x-axis is always _time_, and you are looking at a plot visualizing across two arguably disjoined time ranges (header stamp vs receive time). The points are getting their x-values from `header.stamp` and the gray playback line gets its x-value from the receive time. This has the effect of interleaving the two.

rgov, 2022-05-03:
I guess I understand, and I do see on the Raw Message panel that that the receive time is `1651337130` and the `header.stamp.sec` is `1651337106` (~24 seconds earlier). I was not expecting the clock on the instrument generating these depth messages to be so far off.

defunctzombie, 2022-05-03:
> I was not expecting the clock on the instrument generating these depth messages to be so far off.

Real world problems.

defunctzombie, 2022-05-03:
I'm gonna close this out since the current behavior is by-design. I think there's definitely something more here around visual indication but its not an easy problem to solve because header.stamp could really be in any order so placing the playback marker on a plot with header stamp values is always going to create the possibility of skew like this. I could say that we should never show the playback marker in the plot when there's a header.stamp value plotted but what if you want to plot one series by receive time and another by header.stamp to compare - probably you'd still want to see the playback indicator on the plot?