foxglove/studio #3303: Preserve raw message expansion when switching tabs

ccpowers, 2022-05-03:
If a raw messages panel is inside of a tab panel, and you switch away from that tab, the state of the raw messages panel is not preserved.

You can see this by:
1. Adding a tab panel to your layout
2. Adding a raw message panel in a tab
3. Expanding the message so that the whole header (or other piece of the message) is displayed
4. Click to another tab, and then click back to the raw message panel. The message will no longer be expanded.

Since messages are not expanded by default, this means that one must expand the message every time. I like to create layouts with raw message panels for several topics in a tab panel, but this makes switching between them cost more time.

jtbandes, 2022-05-04:
Given the fixes @defunctzombie made in https://github.com/foxglove/studio/pull/2710, I am surprised this issue occurs in a tab...

defunctzombie, 2022-05-04:
@jtbandes When you switch tabs I believe the mosaic re-mounts all the components. #2710 tracks during the lifetime of the component but not if the component itself is no longer present.

ccpowers, 2022-05-05:
A similar thing happens with other components in tabs, i.e. plot panels, which will re-plot all the data on opening, which can take a while. Probably also caused by re-mounting all components.

foxymiles, 2022-05-09:
Yeah looks like we only render the selected tab so any ephemeral state is going to get lost when switching to a new tab.

https://github.com/foxglove/studio/blob/6e2c8ca905e582d14dfeb95dd351bfc94698ec3d/packages/studio-base/src/panels/Tab/index.tsx#L108-L131

defunctzombie, 2022-05-09:
We could preserve raw message expansion in the layout - but then that means all those changes persist to the layout. We don't have a separate _local_ state concept thats distinct from the saved layout. Given the current architecture of tabs and layouts I'm not sure what the ideal strategy is here. I think we'd need some state that lives neither in a layout nor in the individual component.

foxymiles, 2022-05-09:
I guess we could always render all panels in the tab set and just hide the ones that aren't selected? That would unfortunately mean you take the performance hit for rendering panels that aren't visible.

foxymiles, 2022-05-09:
> We could preserve raw message expansion in the layout - but then that means all those changes persist to the layout. We don't have a separate _local_ state concept thats distinct from the saved [...]

Yeah and this is further complicated in the case of the raw messages panel since the expand/collapse state may really only make sense for a particular message and requires a received message.

defunctzombie, 2022-05-10:
> It's really annoying to re-expand `msg.pose.pose.pose.pose.pose` (a few extra `.pose` added for effect) whenever you want to see something that's a pose and a covariance.

Quick UX question here - have you tried the icon to expand all? Maybe your message has other fields you don't want to expand?

Would it help to auto-expand certain fields? Maybe thats getting too much into the territory of magic rather than just saving what you expanded in the first place.

> I would personally be happy to have the message expansion in the layout :)

👍 thanks for the feedback

ccpowers, 2022-05-12:
Roman: 
 Expand all is nice, but not as nice as keeping the subset expanded. Also, sometimes i'm trying to see things (for instance, headers on laser scans) that are really annoying to expand all on, since the messages may have pieces that are very large / not human interpretable. 

I think auto expanding certain fields could be a good solution. Might actually make sense to go with expanding everything except certain fields, i.e., if the field is too large, don't expand it. I imagine most people want to see most of their fields expanded when they use the raw message panel anyway. Of course, you might end up still with people wanting to preserve expanding or collapsing fields that don't fit that model. 
TL;DR maybe just default to everything being expanded would be a quick fix

 Also, UX confession, I frequently click "toggle diff" instead of "expand all" accidentally, and have not found a good way to make use of that feature.

foxymiles, 2022-05-13:
@ccpowers I've added something along the lines you suggested. In the panel settings there's now an `Auto expand` setting. You can set this to "all" to completely expand all message fields or to "auto" to expand all but the large arrays.

[image]

PR #3360 (2022-05-13): Add new message expansion options in the raw messages panel
**User-Facing Changes**
This adds new auto-expansion behavior to the raw messages panel, giving users more control over when message fields are expanded.

**Description**
This adds a new "auto expand" setting to the raw messages panel with three modes:

1. Off - the current behavior
2. On - all message fields are automatically expanded
3. Auto - message fields under a certain size are expanded (currently this limits arrays to < 5 members)

[image]

Fixes #3303
closed 2022-05-13 by foxymiles