foxglove/studio #3315: Render depth images in 3d panel

amacneil, 2022-05-05:
Several folks have requested the ability to render depth images directly in the 3d panel. Currently we only support this if they are pre-converted into a point cloud, which results in significant amounts of extra data.

I'm not sure how these are typically represented in ROS, so I invite anyone interested in this to comment below with what message type they are using to encode depth images.

clementchef, 2022-05-05:
In the case of realsense cameras, the depthmap are encoded as sensor_msgs/CompressedImage compressed in png (lossless). Each pixel is a 16bit depth in mm see https://www.ros.org/reps/rep-0118.html#id4 OpenNI representation. 
0 value pixel means undefined. 
The conversion to pointcloud requires the depth_image, the camera_info to create the directional vectors (computed once) and optionally the rgb image to color the point cloud. In that case you would assume the depth map and rgb pixel are aligned. 
Creating the pointcloud for that depthmap is done by multiplying the direction vector map by the depthimage and optionally coloring that point with the rgb value for the same pixel.

jaggiK, 2022-05-05:
@amacneil  This would be a great feature to have.
Here are the message types:
depth image : sensor_msgs/Image (16 bit)
camera info:  sensor_msgs/CameraInfo
for color data: sensor_msgs/Image (usually 8 bit)
Happy to help if you need more information in developing this useful feature.

kantengri, 2022-06-26:
ability to visualize any rgb image as a ground layer would be useful to display a map. Seems it is possible to extend OccupancyGrid renderer to accept images.