foxglove/studio #3318: Foxglove Studio (1.9.1) Crashes - "Array buffer allocation failed"

TeeJayR, 2022-05-05:
When I load my ROS bag (duration ~6 min, ~5GB) into Foxglove Studio (version 1.9.1) I reproducibly get an "Array buffer allocation failed" when plotting a signal in a single plot widget. Sometimes the whole GUI just freezes without even showing the error. The signal I want to plot has a frequency of ~25 Hz (>9000 messages).

Is that a known issue? Any hint on how this can be traced down further?

TeeJayR, 2022-05-05:
@foxymiles Ah, sorry, forgot that info: Ubuntu 20.04.

foxymiles, 2022-05-05:
Thanks for the extra details. Can you also please provide a bit more detail on the type of message you're recording and plotting?

TeeJayR, 2022-05-05:
It is a custom ROS message, but in the end I select a `float32` to be plotted over time.

TeeJayR, 2022-05-05:
Unfortunately I am not allowed to share the exact same bag.

In order to isolate the problem I tried to filter the bag for only the topic I want to plot (which causes the crash in the original bag), but this does not reproduce the error (eventhough the bag still has the same length and also uses the same compression, which is lz4 in my case).

I also tried to decompress the original bag, but the crash is then still reproducible, so the compression seems not to be the problem.

So overall, since I noticed plotting the same topic in the (much smaller) filtered bag is super fast compared to plotting only that signal in the original bag, I think the overall number of topics and messages (a total of 809640 messages on 82 topics) is the problem for the app.

foxymiles, 2022-05-05:
Thanks for the extra info. 

One more question. If you go into your settings is "Experimental Bag Player" turned on? If so can you please try turning it off then reloading the app and trying your bag again?

[image]

TeeJayR, 2022-05-05:
The experimental bag player (and all other experimental options) are turned off. Activating it also does not make a difference - the app still crashes.

foxymiles, 2022-05-05:
Ok thanks for running that extra test.

TeeJayR, 2022-05-05:
Hm, after I activated the experimental bag player once and deactivated it again the behavior is different, and the app does not crash anymore. Seems the GUI checkbox was initially just shown deactivated, but the experimental bag player was indeed turned on? I can now reproduce, that the crash is only happening with the experimental bag player.

Without the experimental bag player, the preloading/cashing of the ROS bag stops at some point, further data points are only loaded/plotted when the time slider comes close to the end of the cashed interval.

With the experimental bag player turned on, the preloading/cashing proceeds until the app crashes.

TeeJayR, 2022-05-05:
One more thing I found: If I cancel the inital welcome dialog to go to the settings, and I change the experimental bag player checkbox state there, the "+" button of the "Data source" dialog disappears. So basically, after toggling the state of the experimental bag player, I have to restart the app. Is that known?

TeeJayR, 2022-05-05:
As far as I can see the default behavior of the experimental bag player changed here: https://github.com/foxglove/studio/pull/3249 I am not familiar with the code base (yet?!), but maybe the GUI does not read the correct initial state of the checkboxes after a fresh install?

foxymiles, 2022-05-05:
Yes you're right we recently changed the experimental bag player to be the default. 

So far I'm not seeing the issue you mention with the `+` disappearing. You're using the linux desktop client right?

TeeJayR, 2022-05-05:
Correct.

defunctzombie, 2022-05-07:
I have a bag file where I am able to reproduce Array Buffer Allocation failed. So far I've not been able to track down the root cause. It only happens with the experimental bag player. I originally thought it was a leak but that hasn't turned out to be the case in my findings _so far_. I've been working on logic changes to the experimental bag player which seem to resolve the issue by using a different strategy for handling some internal logic. It is unfortunate I cannot find the root cause of failing to allocate memory - it might be a browser memory use issue with many repeated allocations.

foxymiles, 2022-05-10:
@TeeJayR If you try loading your bag into the preview deployment here do you get the same error? https://studio-7cdp26sdi-foxglove.vercel.app/

TeeJayR, 2022-05-11:
Unfortunately I can not open the preview version remotely due to information protection reasons. Any chance I can download a standalone/desktop version of the preview?

foxymiles, 2022-05-11:
@TeeJayR I think we've identified the problem but are still discussing the best fix. I think we'll have a new desktop release out with the fix very shortly.

defunctzombie, 2022-06-30:
@TeeJayR circling back on this. I recently merged some changes to the experimental bag player which should resolve this issue. These are available on the web version now and will be in next week's desktop release. Please give that a try. I am closing this issue because I believe it is fixed by those changes. If you discover otherwise please re-open with any new details or steps to reproduce.

closed 2022-06-30 by defunctzombie

TeeJayR, 2022-07-22:
Tested with 1.18.0 and does not seem to occur anymore. Thanks!