foxglove/studio #3338: Rename Node Playground

amacneil, 2022-05-07:
"Node Playground" is not particularly descriptive, and many users are surprised to learn what it does. The term "Node" is also confusing, because Node Playground scripts are not published back to the data source (and we have no plans to support that).

Some suggestions have been put forward:
* User Scripts
* User Functions
* Function Editor
* Live Functions
* Data Transforms (potentially confusing with 3D transforms)

More suggestions are welcome.

defunctzombie, 2022-05-08:
* Scripts

defunctzombie, 2022-05-08:
Function could be interesting but might also be something we want to use with custom plot functions?

* Message Transformer
* Message Scripts
* Topic Scripts

amacneil, 2022-05-08:
Do you think we will have a concept of custom plot functions separate from node playground? Or just make it easier to define an inline NP script to generate plot series?