foxglove/studio #3344: Failure to connect to ROS 1 live connection to a ROS instance on another computer

jackiescanlon, 2022-05-10:
**Description**
Starting on May 6th, when attempting to connect to a ROS instance on another machine, my team is experiencing this issue:

[image]

No data will come through, but it will show me all the topic names available.
This does not happen when using Foxglove on the same machine that ROS is running on.

- Version: I have tested this on 1.10.1., 1.10.0, 1.9.1, and 1.7.0. Doesn't seem to have any effect on this issue.
- OS: Ubuntu 20.04

**Steps To Reproduce**

1. Start a ROS instance on a separate computer that you can connect to
2. Open Foxglove studio and use Live Connection - ROS 1 as the connection type
3. Change the `ROS_MASTER_URI` from `http://localhost:11311` to `http://<IP of ROS computer>:11311`
4. Attempt to read data from a topic in any panel

**Expected Behavior**
Data is displayed as expected.  This is what used to happen, following these same steps, prior to May 6.

**Actual Behavior**
5. In the Data Source > Problems panel, a "ROS node error" appears
6. Clicking on the error I receive this:
[image]

jhurliman, 2022-05-10:
Can you resolve the displayed hostname to an IP address on the computer where Studio is running?

jackiescanlon, 2022-05-11:
I can, but only using the alias I gave to it.

Your question led me to the answer. Basically, in my `/etc/hosts` file, I was calling that machine `sitl`. That machine knows itself as `mfcs-sitl` though. In Foxglove, I used to be able to connect to it via a `ROS_MASTER_URI` of `http://sitl:11311`, or `http://<true ip address>:11311`. All of sudden, that stopped working.

Now, *neither* a `ROS_MASTER_URI` of `http://sitl:11311`, or `http://<true ip address>:11311` work. Instead, I had to change my `/etc/hosts` file to exactly match the true name of the machine `mfcs-sitl`. Now I can access that machine both with a `ROS_MASTER_URI` of `http://mfcs-sitl:11311`, or `http://<true ip address>:11311`

 It must've been a change in an underlying ROS library and not a change in Foxglove, because like I said I tested a bunch of different versions of Foxglove. This is a super annoying nuance but one my team can easily live with. Closing this issue.

closed 2022-05-11 by jackiescanlon

soumith, 2022-05-12:
I think I'm seeing the same issue, but I'm not at the luxury of changing my `/etc/hosts` file.

Is there a way we can make `http://<true ip address>:11311` work again?

soumith, 2022-05-12:
I think my issue is worse. It doesn't even display the list of topics, it just spins / is stuck like this.

[image]

jhurliman, 2022-05-12:
> It must've been a change in an underlying ROS library and not a change in Foxglove, because like I said I tested a bunch of different versions of Foxglove.

Studio packages its [own implementation](https://github.com/foxglove/ros1) of ROS1 comms, there are no shared libraries in use outside of the different version of Studio. It sounds like a networking change in the computer(s) running Studio or the network they're running on. Not being able to establish a ROS1 connection to https://<roscore_ip_address:11311/ is definitely a red flag. Can you curl that URL from the computer running Studio? It should return a 404 or some other HTTP status code.

jackiescanlon, 2022-06-21:
I'm sorry it took me so long to get back to you, it took me awhile to debug it. But I figured out what changed - you were completely right, it was a configuration on our end that I didn't realize was different.

On one robot we export a `ROS_IP` environment variable, and on the one I was having problems with, we don't. I was able to replicate my issues. Here's what I found, using an example robot where:
* The IP of the robot is 11.22.33.44
* The full hostname that the robot knows itself as is `abcdef`
* The nickname I would prefer to give it is `ab` (in this example its only 4 characters shorter than the full name, but in real life my robot has a more complicated hostname that I prefer not to type out every time I want to connect to it via Foxglove)

[image]

Generally, when using an exported `ROS_IP`, the user has a lot of options to connect, including using a nickname in the `/etc/hosts` file. But when not setting `ROS_IP`, those options are greatly limited. The squares I highlighted in yellow represent cases that I am really confused about why they don't work. 

Regardless... now we know the secret sauce, which is to set `ROS_IP=11.22.33.44` (using the real IP address of course) on each robot. This will give my team a variety of options to connect in Foxglove.

Thanks for your help @jhurliman!

@soumith, maybe you have the ability to set the `ROS_IP` on your robot and that could fix your issue for you?