foxglove/studio #3353: Displaying a map from disk in 3d panel

marijapili, 2022-05-12:
A useful option would be the possibility to load a static map from disk to display in the 3d panel, so that we don't have to record the OccupancyGrid to the bag file to be able to visualize it.

A format similar to the one used for [http://wiki.ros.org/map_server](map_server) would be great.

defunctzombie, 2022-05-12:
How would you want to identify what version of the map to load? What if the user does not have the maps on their local computer?

marijapili, 2022-05-12:
The idea is that the user can define which file to load - for example, with the map_server, the user passes a .yaml file containing all the necessary data as an argument. So something like that, loading the desired map into Foxglove.

That is useful when repeatedly visualizing data from the robot that is interacting with the same static environment.

defunctzombie, 2022-05-14:
> same static environment.

What happens when that static environment changes over time? In my experience even mapped static environments undergo some changes (i.e. re-mapping).

marijapili, 2022-05-17:
The map is then reloaded manually. It makes sense in our environment to avoid saving the occupancy grid topic in every one of the many bags we are recording for analysis, it helps us reduce the size of the bag

defunctzombie, 2022-06-08:
Right now this is not likely something we would introduce for messages like OccupancyGrid. We prefer that data is loaded into Studio through the data source so the data comes from one known source and is available regardless of your computer setup. Something we've seen users do in this case is record occupancy grids at a much slower rate (like 5 minutes, 10 minutes, etc). So the data is still recorded and available.