foxglove/studio #3369: Feature Request - Publish Point / Publish Pose Persist with Saved Layout

liamreneroy, 2022-05-14:
Feature Request - Publish Point / Publish Pose Persist with Saved Layout

Hi guys, I think this would be a nice UX feature that is likely a super easy thing to do.

In the 3D layout, you can hold right-click the "publish point" and change it to "publish pose". 
See attached photo.

As of now, if you change it from pub point to pub pose, and then save the layout, when you re-open FloxGlove studio, your selection reverts back to pub point. I need to change it back to pub pose every time I open the software. 

It would be cool to have it if you saved a layout with Publish Pose enabled, that the selection persists the next time you open the software. 

- Version: 1.10.1
- OS: Ubuntu 20.0
- 
[image]
4

foxymiles, 2022-07-25:
@jtbandes It sounds like the new3d panel does remember the tool selection right?

jtbandes, 2022-08-01:
Yes, this was done in https://github.com/foxglove/studio/pull/3837

closed 2022-08-01 by jtbandes